keesverruijt/BR24radar_pi #64: please tag releases

mschiff, 2015-01-11:
Hi!

It would be very usefull to have tagged releases, so the source for a specific version of th eplugin can be downloaded. Please add a release/tag for your released versions. This makes packaging a lot easier. 

I'd like to package this plugin for Gentoo Linux and for that it requires a reliable specific download URL for the releases. 

Thanks in advance!

canboat, 2015-01-12:
Hi @mschiff -- good idea!

I've tagged v1.1 and might even start creating releases (with binaries) for v1.2. I don't really know whether there are any advantages to the latter over how I publish them now, maybe we should just do that as it seems the official thing to do.

Is the tag good enough for your purposes?

mschiff, 2015-01-12:
Thanks. Looks good! As long as I can offer a download link to our users like this "https://github.com/canboat/BR24radar_pi/archive/v1.1.tar.gz" everything is fine.

Thanks again!

closed 2015-05-17 by canboat