keesverruijt/BR24radar_pi #65: Make use of "heading on radar" when available

douwefokkema, 2015-02-13:
When an NMEA2000 signal with heading is connected to the RI10 interface box, the radar will include the current heading in every scan-line. Making use of this feature would improve the mapping of the radar image on the chart. A practical benefit is that users with an NMEA2000 system would no longer need to use a converter to provide OCPN with an NMEA183 heading.

I have made a prototype version of this functionality that works fine with me (only tested for 4G radar), available on request (without support). I could publish it on github if I knew how.

NAHANNIV, 2015-03-11:
I'm interested in this. I was not planning to install the RI10, even though I got one with the 4G RADAR.
Three questions: 
1/ How is the mapping of the RADAR image improved ?
2/ Are you passing the heading data into OpenCPN from the plugin, what about when the RADAR is in STANDBY or OFF ?
3/ Can MARPA be easily implemented ? It looks like TTM and TLL are already implemented in OpenCPN, seems that only Acquire Target and Cancel MARPA Target Tracking would need to be added to the plugin.

douwefokkema, 2015-03-11:
I have not yet been able to test this with moving ship and moving 
targets, my boat is still on the hard. But in theory:

1. With heading_on_radar a target will be mapped on the chart based on 
the bearing which is calculated using the heading of the moment of 
scanning. Currently the bearing of the targets is calculated using the 
heading at the moment of displaying. With a turning ship, a target could 
in theory be displayed in different locations with consecutive display 
operations. A full radar scan takes 2.5 seconds, OCPN lets the radar_pi 
display the image every second. So I expect the image to become more 
stable especially in a seaway when heading_on_radar is used.

2. Yes, the heading received from the radar is passed to OCPN with 
highest priority. As this data comes in over the ethernet included in 
the scan data, it will drop away with radar in standby or off. If 
another heading source is available (NMEA) OCPN will fallback on that, 
otherwise do without.

3. I have not yet looked into Marpa. I don't care too much about it as 
most important targets carry AIS. Yet it could be interesting to analyze 
the radar data stream with Marpa ON and see what happens.

NAHANNIV, 2015-03-12:
I think something like this was implemented in the Garmin RADAR plugin V4.1 (see point 2):
http://www.cruisersforum.com/forums/f134/garmin-radar-plugin-version-1-4-released-122446.html#post1491595

Could you add the Heading info to the scan packets when received in the plugin for people who don't have the RI10 installed ? How much latency would there be at that point ?

Regarding using the Heading from the RADAR, I would not want to have to wait for a fail over every time I turn off the RADAR. Could the priority be configurable. or the function be selectable ? Perhaps automatically disabled if Heading is already being received ?

If the MARPA Acquire Target messages are also on NMEA2000 it will be a while before that could be implemented.

Cheers,
JM.

douwefokkema, 2015-03-12:
Yes, that is what I do in my experimental version of the pi right now. If heading_on_radar is not available I take the heading that the pi gets from the NMEA stream and use that for mapping the particular block of data. I do not know with how much latency this heading is passed from the NMEA to the pi. The effect of this can be seen in the attached screen movies. These are made by having TCPreplay playing a recorded radar data stream (without heading_on_radar). The NMEA is made by a simulator that delivers a rotating heading of 1 degree / sec. The effect of this is that the radar image will turn (as the ship was not turning when the recording was made). In the first movie (normalradar.wmv) the whole image is rotated with each display cycle, including the targets that have not been rescanned yet. In the second movie (directheading.wmv) the NMEA heading is linked with the data blocks at the moment of receiving the data. Now only the parts of the image that have been rescanned since last display will rotate.
The last movie (fastradar.wmv) is the same pi as the second, but now I have adapted OCPN such that it will display 5 times per second. Here you get the feeling of a rotating beam.

Regarding passing back the heading_on_radar to OCPN, this could be left out, not essential. But normally speaking it should be the same heading as on the NMEA. So if it is passed to OCPN and it falls away, OCPN will smoothly switch over to the other heading source.

Regarding the movies, I can't attach them, file type not supported. So I put them on YouTube:
normalradar : http://youtu.be/Waaglug23C8
directheading: http://youtu.be/QTGONuX_-mw
fastradar: http://youtu.be/IdbsxDCpJQY

NAHANNIV, 2015-03-12:
I was not thinking so much of the latency of the HDG received by the plugin, but, how much difference there would be with a scan packet having heading recorded in the scanner vs one that gets heading recorded when received by the plugin; I.E. Is it worth having the extra RI10 hardware ?

What sort of hardware are you testing on ? I wonder if the 5HZ version would work on an ARM board.

Perhaps the simplest thing to do with passing the heading back is to make it a lower priority so that the user can choose. I don't know what the failover timeout is, but there would be a lapse before switching every time the radar is turned off.

douwefokkema, 2015-03-13:
Latency on the side of incoming data should not be much of a problem. 
With heading_on_radar a heading gets connected to each scanline. Without 
heading_on_radar a heading is linked with one UDP frame of 32 scanlines. 
These frames are coming in at a rate of about 25 frames per second.

NAHANNIV, 2015-04-16:
Are you able to add the changes you made to the br24 code here on github ?
Also there have been some changes made by Hakan to implement timed transmit.

Also, can the changes to allow faster update rates be added to OpenCPN ?

I would like to test it.

Thanks,
JM.

douwefokkema, 2015-04-17:
Hi NAHANNIV,
Currently my changes are on https://github.com/douwefokkema/BR24radar_pi/tree/experiment. 
This version however is not perfect, Hakan finds problems loosing the position in the pi. On my systems it runs fine.
Main differences with the released version:
- making use of the heading on radar
- when heading on radar is not available use heading at time of reception of data
- timed transmit (contribution of Hakan)
- faster refresh of the radar image (adjustable with the refresh control)
- performance improvement of the display process to allow the faster refresh
- more accurate range settings (identical commands as HDS display)
We have the intention to incorporate (most of) this in a released version when stable. But as said, problems have been reported and I have not been able to solve these yet.
You are welcome to test. If you need a compiled dll mail me on df@percussion.nl.

rgleason, 2015-04-17:
Seems to me that with timed transmit, the heading and some adjustment to persistance on the opencpn side, the amont of data could be reduced enough so that a wifi connection might work.

With some data files from either radar you could do a trial over a wifi router with two computers. This would imform  if there changes needed or if it would work at all.

I recall a discussion thread about what  wifi ethernet protocal should be used because this is all one way and wifi breaks up the signal for two way as I ubderstand it.  Still the setup needs a thin pipe for acknowledge and a fat pipe for radar data. Anyway maybe we aren't there yet. 

However Furuno has ads all over the place for radar on your tablet via built in wifi in the radar itself.  Who is next?

If there was a radar file available i might try it.

rgleason, 2015-04-17:
Muttinick was in the protocal discussion with several others. It appears Furuno has solved this programming protocal problem.

rgleason, 2015-04-17:
Maybr ocpn will get a furuno radar plugin?  Woold be nice but I would rather a broadband for power sipping.

canboat, 2015-04-17:
Rick, please check where you are posting this. This is ending up in issue #65 on github, which is probably not what you intended. Please either open a new issue if you have an actual issue with the BR24 plugin, or comment somewhere else if you just want to talk.

rgleason, 2015-04-17:
Maybe Furuno. Is processing the data at the radar and sending a compressed image via onboard wifi? Instead of nmea?

rgleason, 2015-04-17:
Perhaps As vector since that is smaller.

rgleason, 2015-04-17:
Whoops. Sorry. On my phone cant sleep.

canboat, 2015-05-17:
Fixed in v1.2

closed 2015-05-17 by canboat