keesverruijt/BR24radar_pi #67: HDG data requirement

syspringdawn, 2015-04-17:
I note that in the current release versions of the PI the presence of an HDG input is a hardcoded pre-req for operation. Whilst I understand the desirability of that its not always easy to achieve, for reasons of wiring or just cost. Could the presence of HDG be made user selectable by way of an ini parameter or similar mechanism please ?

canboat, 2015-04-17:
Sorry, no can do. The only presentation mode is radar overlay as there is no separate radar display on a different screen. As overlay will only work if the radar data is rotated correctly so that it matches the chart, this is not possible without some heading input.

On the plus side, the next release will accept heading data coming from the RI-10/11 interface box, so if you wire that up to a compass it will work -- in fact, it will work better than data going into OpenCPN.

If you can afford a broadband radar, you can afford a compass as well.

closed 2015-04-17 by canboat

syspringdawn, 2015-04-17:
I already can wire up the RI-10 to HDG output from a Raymarine AP, its a case of getting the wires there through existing trunking thats the hard bit. I also have an NSS8 and find that outputs NMEA data through an IP port that I can feed to the boats network. I anticipate that once I get the wiring done I can get the HDG data in addition to the GPS etc data that already comes from the NSS8, its just not that high a priority. Guess I'll be using the older version of the PI for a bit longer then, pity.

canboat, 2015-04-17:
Not sure how you use it then except if you use the 'feature' where the chart is painted black so you only see the radar returns. Guess you also don't have AIS, as those would then end up in a different rotation.

If you already use the GPS from the NSS8 I don't see why you can't use the NSS NMEA stream for compass data.

But feel free to fork the code and butcher whatever you want. This is open source after all.

douwefokkema, 2015-04-17:
You will know of course, but just to be complete: HDM is sufficient if you use the WMM plugin for variation.

syspringdawn, 2015-04-17:
Apologies, gentlemen. I will have to retest when I get back to the boat as I note that I have the HDx inputs disabled for the NSS line in my OCPN configuration (cant remember why I did that but it must have been last year). Perhaps my confusion is that the NSS wont allow overlay without a 10Hz HD? input signal. I intend to derive that from an AT10HD connected between the Raymarine AP and the Simnet bus. It seems from your descriptions that the current PI would accept HDx sentences from the NSS as acceptable input for overlay in OCPN without the 10Hz signal?

As an aside I dont currently have AIS into the NSS, only into OCPN via a serial/USB interface either direct or via kplex into the network.

canboat, 2015-04-17:
> Perhaps my confusion is that the NSS wont allow overlay without a 10Hz HD? input signal. I intend to derive that from an AT10HD connected between the Raymarine AP and the Simnet bus. 

All Navico MFDs require the high speed heading data to originate from the radome. This can be done by using a NMEA 2000 compass and an RI-10 interface box. You can use a AT10HD to convert NMEA0183 heading data to NMEA2000 but the heading data needs to be at 10 Hz on the NMEA0183 stream. Not sure whether your Raymarine AP will send data that often.

An alternate to the RI-10 + AT10HD is to use a RI-11, which accepts NMEA0183 data directly -- but again it needs to be 10 Hz.

> It seems from your descriptions that the current PI would accept HDx sentences from the NSS as acceptable input for overlay in OCPN without the 10Hz signal?

Correct, the plugin accepts 1 Hz or lower heading input. Obviously, the lower the quality of the heading data the lower the quality of the overlay rotation and matching.

The new changes being made by Douwe, which will be in the next release, prefer to use the heading data sent out by the radar if present. This improves the radar image overlay, especially during fast heading changes of the boat.

syspringdawn, 2015-04-17:
"..You can use a AT10HD to convert NMEA0183 heading data to NMEA2000 but the heading data needs to be at 10 Hz on the NMEA0183 stream. Not sure whether your Raymarine AP will send data that often.."

Its an S3G, the spec says it *should*. I know it outputs heading data as thats been tested into my laptop, I cant remember if OCPN knew what the update frequency was. 

The use of an AT10 is consistent with the install manual for the radar head, presumably the RI-10 picks up the heading signal from Simnet and feeds that to the radar head. Presumably the PI changes require an RI-10 to get heading data from the radar ? iirc, the RI-10 was part of the 'kit' in Europe but not in the US, there was discussion on the forum - I think - about using the radar head as a 'standalone' feed to OCPN.