keesverruijt/BR24radar_pi #69: NMEA parser can crash on VDR data

canboat, 2015-05-22:
When using the VDR data recorder the plugin is reported to crash when compiled on Ubuntu.

VDR data:

    $AIHDT,045.0,T
    $GPGGA,174708.000,4311.8478,N,07934.4941,W,1,07,1.5,82.4,M,-34.7,M,,0000*56    
    $GPGSA,M,3,26,15,29,13,21,05,18,,,,,,3.6,1.5,3.3*39
    $GPRMC,174708.000,A,4311.8478,N,07934.4941,W,0.00,045.00,151214,,,A*7B

Crash dump:

    *** opencpn (wxWidgets 2.8.12) crashed ***, see backtrace!
    0x83577f6 in wxCrashPrint::Report() at ??:?
    0x8142324 in MyApp::OnFatalException() at ??:?
    0xb76679f3 in wxFatalSignalHandler at ??:0
    0xb76f7400 in [0xb76f7400] at ??:0
    0xb7629727 in wxStringBase::operator=(wxStringBase const&) at ??:0
    0x8162062 in wxString::operator=(wxString const&) at ??:?
    0x836e544 in NMEA0183::PreParse() at ??:?
    0xb2feb134 in NMEA0183::Parse() at ??:0
    0xb2fdcbbe in br24radar_pi::SetNMEASentence(wxString&) at ??:0
    0x82e5032 in PlugInManager::SendNMEASentenceToAllPlugIns(wxString const&) at ??:?
    0x8323868 in Multiplexer::OnEvtStream(OCPN_DataStreamEvent&) at ??:?
    0xb75d3df6 in wxAppConsole::HandleEvent(wxEvtHandler*, void (wxEvtHandler::*)(wxEvent&),     wxEvent&) const at ??:0
    0xb76646ae in wxEvtHandler::ProcessEventIfMatches(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) at ??:0
    0xb7664a49 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) at ??:0
    0xb7664b0b in wxEvtHandler::ProcessEvent(wxEvent&) at ??:0
    0xb76645f8 in wxEvtHandler::ProcessPendingEvents() at ??:0
    0xb75d41a9 in wxAppConsole::ProcessPendingEvents() at ??:0
    0xb740b340 in wxAppBase::ProcessIdle() at ??:0
    0xb7376022 in  at ??:0
    0xb6579c50 in  at ??:0
    0xb657d0a7 in g_main_context_dispatch at ??:0
    0xb657d468 in  at ??:0
    0xb657d76b in g_main_loop_run at ??:0
    0xb68df450 in gtk_main at ??:0
    0xb738c17e in wxEventLoop::Run() at ??:0
    0xb740b14f in wxAppBase::MainLoop() at ??:0
    0xb740ac4f in wxAppBase::OnRun() at ??:0
    0xb7608d16 in wxEntry(int&, wchar_t**) at ??:0
    0xb7608db6 in wxEntry(int&, char**) at ??:0
    0x813d0c4 in main at ??:?
    Aborted

douwefokkema, 2015-05-22:
I tried to reproduce this with OCPN 4.0 and wxWidgets 2.8.12 on Windows, no crash, runs fine. 
Same on Ubuntu OCPN4.1 with wxWidgets 3.0, running fine, also with radar (TCPreplay). Currently I do not have the Ubuntu with wxWidgets 2.8.12 installed so I cannot test that.
In these tests I used and a VDR file repeatedly containing the mentioned lines of NMEA output.

douwefokkema, 2015-05-22:
I would like to have a look at the opencpn.conf file for this case. When testing under Ubuntu earlier most problems were related with the initializations.

ptulp, 2015-05-22:
I have seen a similar problem with the logbook plugin (see CF Logbook Konni #664)
As far as I could determine the Parse function in the plugin did call the PreParse
in OpenPCN/src/ PreParse i.s.o. the PreParse in the plugin.
As I could not find the real reason for this I crudely solved it by renaming the NMEA0183 definitions
to NMEA0183L in the plugin. 
The problem was not seen after that. 
I am not very good at C++ so may be there is a better way to fix this as it can be seen for all plugins using NMEA I guess.

NAHANNIV, 2015-05-22:
Here is my config file:

[PlugIns]
[PlugIns/GRIB]
bEnabled=1
[PlugIns/Dashboard]
bEnabled=1
Version=2
FontTitle=sans Italic 10
FontData=sans 14
FontLabel=sans 8
FontSmall=sans 8
DashboardCount=1
SpeedometerMax=12
SpeedUnit=0
DepthUnit=3
DistanceUnit=0
WindSpeedUnit=0
[PlugIns/Dashboard/Dashboard1]
Name=13722a40-8bee-486b-8d33-535a70b21439
Caption=Dashboard
Orientation=H
InstrumentCount=3
Instrument1=20
Instrument2=7
Instrument3=33
[PlugIns/Celestial]
bEnabled=1
[PlugIns/FindIt]
bEnabled=1
[PlugIns/VDR]
bEnabled=1
InputFilename=/home/nahanni/Desktop/Nema Stream.txt
OutputFilename=
Interval=1000
[PlugIns/Climatology]
bEnabled=1
[PlugIns/Logbook]
bEnabled=1
[PlugIns/WeatherFax]
bEnabled=1
[PlugIns/AIS\ Radar\ view]
bEnabled=1
[PlugIns/SAR]
bEnabled=1
[PlugIns/WMM]
bEnabled=1
[PlugIns/ROUTE]
bEnabled=1
[PlugIns/Calculator]
bEnabled=1
[PlugIns/rtlsdr]
bEnabled=1
[PlugIns/libSAR_i386_pi18_v08_pi.so]
bEnabled=0
[PlugIns/liblogbookkonni_i386_pi16_v11_pi.so]
bEnabled=0
[PlugIns/libclimatology_i386_pi19_v02_pi.so]
bEnabled=0
[PlugIns/libdashboard_pi.so]
bEnabled=1
[PlugIns/libroute_i386_pi18_v10_pi.so]
bEnabled=0
[PlugIns/libwmm_pi18_v07a_i386_pi.so]
bEnabled=0
[PlugIns/libaisradar_i386_pi17_v096_pi.so]
bEnabled=0
[PlugIns/libfindit_i386_pi16_v10_pi.so]
bEnabled=0
[PlugIns/libweatherfax_pi18_v06_i386_pi.so]
[... 552 lines cut ...]
en-cb4437f1=ToolTips:Ubuntu 11:rgb(0, 0, 0)
en-dc3cb5f7=Marks:Ubuntu 12:rgb(0, 0, 0)
en-df0f7403=CurrentValue:Ubuntu 11:rgb(0, 0, 0)
en-e441042a=ExtendedTideIcon:Ubuntu 11:rgb(0, 0, 0)
en-fbbaefb3=TideCurrentGraphRollover:Ubuntu 11:rgb(0, 0, 0)
en_US-3582063a=ChartTexts:Ubuntu 12:rgb(0, 0, 0)
en_US-6052cba4=Console Value:Ubuntu 12:rgb(0, 0, 0)
en_US-68c1d6c4=Console Legend:Ubuntu 12:rgb(0, 0, 0)
en_US-b25a3899=StatusBar:Ubuntu 12:rgb(0, 0, 0)
en_US-cb4437f1=ToolTips:Ubuntu 12:rgb(0, 0, 0)
[Directories]
S57DataLocation=
SENCFileLocation=
InitChartDir=/home/Charts
GPXIODir=
TCDataDir=/usr/share/opencpn/tcdata/europe
WMMDataLocation=//plugins
WeatherFaxDataLocation=//plugins
[AUI]
AUIPerspective=layout2|name=ChartCanvas;caption=;state=768;dir=5;layer=0;row=0;pos=0;prop=100000;bestw=1400;besth=968;minw=-1;minh=-1;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=13722a40-8bee-486b-8d33-535a70b21439;caption=Dashboard;state=2099187;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=450;besth=167;minw=450;minh=167;maxw=-1;maxh=-1;floatx=3;floaty=601;floatw=726;floath=365|dock_size(5,0,0)=285|
[Groups]
GroupCount=0
[TideCurrentDataSources]
tcds0=/usr/share/opencpn/tcdata/HARMONIC.IDX
tcds1=/usr/share/opencpn/tcdata/europe/HARMONIC.IDX

NAHANNIV, 2015-05-22:
Here is my logfile up to the crash:

08:28:55 EDT: 2015-05-22
08:28:55 EDT:  ------- Starting OpenCPN -------
08:28:55 EDT: Version 4.0.0 Build 2015-01-08
08:28:55 EDT: wxWidgets version: wxWidgets 2.8.12
08:28:55 EDT: MemoryStatus:  mem_total: 495 mb,  mem_initial: 9 mb
08:28:55 EDT: SData_Locn is /usr/share/opencpn/
08:28:55 EDT: Using existing Config_File: /home/nahanni/.opencpn/opencpn.conf
08:28:55 EDT: Styles loading from /usr/share/opencpn/uidata/styles.xml
08:28:55 EDT: No styles found at: /home/nahanni/
08:28:55 EDT: No styles found at: /home/nahanni/.opencpn/
08:28:55 EDT: Detected display size: 370 mm
08:28:55 EDT: Detected display size: 370 mm
08:28:55 EDT: Setting Viewpoint Lat/Lon 43.1975, -79.5749
08:28:55 EDT: Setting Ownship Lat/Lon 43.1975, -79.5749
08:28:55 EDT: Loading navobjects from navobj.xml
08:28:55 EDT: Done loading navobjects
08:28:55 EDT: System default Language:  en_US
08:28:55 EDT: Opencpn language set to:  en_GB
08:28:56 EDT: ChartSymbols loaded from /usr/share/opencpn/s57data/chartsymbols.xml
08:28:56 EDT: Using s57data in /usr/share/opencpn/s57data
08:28:56 EDT: Setting Viewpoint Lat/Lon 43.1975, -79.5749
08:28:56 EDT: Setting Ownship Lat/Lon 43.1975, -79.5749
08:28:56 EDT: Creating glChartCanvas
08:28:56 EDT: PlugInManager searching for PlugIns in location /usr/lib/opencpn
08:28:56 EDT: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libvdr_pi.so
08:28:56 EDT:   /usr/lib/opencpn/libvdr_pi.so
              API Version detected: 106
              PlugIn Version detected: 3
08:28:56 EDT: Opencpn language reload for:  en_GB
08:28:56 EDT: Loading catalog for:  opencpn-vdr_pi
08:28:56 EDT: Loading catalog for opencpn core.
08:28:56 EDT: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libdashboard_pi.so
08:28:56 EDT:   /usr/lib/opencpn/libdashboard_pi.so
              API Version detected: 106
              PlugIn Version detected: 102
08:28:56 EDT: Opencpn language reload for:  en_GB
08:28:56 EDT: Loading catalog for:  opencpn-vdr_pi
08:28:56 EDT: Loading catalog for:  opencpn-dashboard_pi
08:28:56 EDT: Loading catalog for opencpn core.
08:28:56 EDT: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libbr24radar_pi.so
08:28:56 EDT:   /usr/lib/opencpn/libbr24radar_pi.so
              API Version detected: 110
              PlugIn Version detected: 102
08:28:56 EDT: Opencpn language reload for:  en_GB
08:28:56 EDT: Loading catalog for:  opencpn-vdr_pi
08:28:56 EDT: Loading catalog for:  opencpn-dashboard_pi
08:28:56 EDT: Loading catalog for:  opencpn-br24radar_pi
08:28:56 EDT: Loading catalog for opencpn core.
08:28:56 EDT: BR24radar_pi: Configuration file values initialised
08:28:56 EDT: BR24radar_pi: Log verbosity = 0 (to modify, set VerboseLog to 0..4)
08:28:56 EDT: BR24radar_pi: GuardZone 1: Off
08:28:56 EDT: BR24radar_pi: GuardZone 2: Off
08:28:56 EDT: OpenGL-> Renderer String: Gallium 0.4 on ATI RV350
08:28:56 EDT: OpenGL-> Version reported:  2.1 Mesa 10.1.3
08:28:56 EDT: OpenGL-> Texture rectangle format: de1
08:28:56 EDT: OpenGL-> glGenerateMipmap unavailable
08:28:56 EDT: OpenGL-> Using Vertexbuffer Objects
08:28:56 EDT: OpenGL-> Using Framebuffer Objects
08:28:56 EDT: OpenGL-> Using FBO Stencil buffer
08:28:56 EDT: OpenGL-> Using Stencil buffer clipping
08:28:56 EDT: OpenGL-> Using Scissor Clipping
08:28:56 EDT: OpenGL-> Using s3tc dxt1 compression
08:28:56 EDT: OpenGL-> Compressed tile size: 128kb (6:1)
08:28:56 EDT: OpenGL-> Minimum cartographic line width:  1.0
08:28:56 EDT: OpenGL-> Minimum symbol line width:  1.0
08:28:57 EDT: ChartDB Cache policy:  Application target is 242 MBytes
08:28:57 EDT: Loading chart db version: V018
08:28:57 EDT: Chartdb: Chart directory list follows
08:28:57 EDT:   Chart directory #0: /home/Charts/CA_CHTS
08:28:57 EDT:   Chart directory #1: /home/Charts/Cm93 jan 2011
08:28:57 EDT: GPS Watchdog Timeout is: 6 sec.
08:28:57 EDT: Initializing Chart /home/Charts/Cm93 jan 2011
08:28:57 EDT: CM93Composite Chart Root is /home/Charts/Cm93 jan 2011/
08:28:57 EDT: Loaded CM93 Dictionary from /home/Charts/Cm93 jan 2011/
08:28:58 EDT: Loading CM93 cell /home/Charts/Cm93 jan 2011/03900840/E/03990841.E
08:28:58 EDT: Loading CM93 cell /home/Charts/Cm93 jan 2011/03900840/D/03990840.D
08:29:00 EDT:    ***GPS Watchdog timeout at Lat:43.1975   Lon: -79.5749
08:29:12 EDT: Changing NMEA Datasource for GPGGA to  (Priority: 0)
08:29:13 EDT: BR24radar_pi: GPS position is now known
08:29:13 EDT: Changing NMEA Datasource for GPGSA to  (Priority: 0)
08:29:14 EDT: Changing NMEA Datasource for GPRMC to  (Priority: 0)
08:29:15 EDT: BR24radar_pi: Heading source is now COG
08:29:16 EDT: Changing NMEA Datasource for AIHDT to  (Priority: 0)
08:29:17 EDT: BR24radar_pi: Heading source is now HDT
08:29:27 EDT:

NAHANNIV, 2015-05-22:
Perhaps I should just update to latest OpenCPN and wxWidgits 3.0 ?

douwefokkema, 2015-05-22:
I tested with your .conf  / .ini file under Win7 (OCPN4.0), no problems.
On Ubuntu (OCPN4.1 wxWidgets 3.0) it seems to run fine, but I get following error messages when opening the options window after having run the radar_pi with your VDR file:

(opencpn:2430): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17

and on a next run:

(opencpn:2715): Gdk-CRITICAL **: IA__gdk_draw_layout: assertion 'PANGO_IS_LAYOUT (layout)' failed
(opencpn:2715): Pango-CRITICAL **: pango_layout_get_line_count: assertion 'layout != NULL' failed

So there must be something wrong in the radar_pi. Needs more time to investigate.

NAHANNIV, 2015-05-22:
I get the GTK warnings all the time on this UBUNTU laptop and my ARM platform.
I was told by NOHAL that it's normal.

I also have no problem with the current version on Win7.

A previous version of the plugin worked on my UBUNTU laptop; I was able to use it on my boat and see an excellent RADAR image. Then I built latest br24 from source, and have not been able to get it to work.

I think later I will try to update everything to latest versions.

canboat, 2015-05-22:
@NAHANNIV it is probably more helpful if you rebuild both O and the plugin using the debug build. Once you've done this then please upload the backtrace. If compiled with debug this will include line numbers and source files.

This could very well be related to wx 2.8.12 which is why it doesn't show up on newer releases.

canboat, 2015-05-23:
@NAHANNIV, commit c68abc3 implements @ptulp 's suggestion. Can you try again? Note that this is NOT in the Beta released earlier today.

NAHANNIV, 2015-05-25:
I was at the boat this weekend, eager to try the RADAR, so I updated to the latest of everything before I got the message from canboat about a special version: OpenCPN 4.1.427, wxWidgets 3.0, br25_pi.

I still had wxWidgets (3.0) crashing.

Then I noticed that OpenCPN 4.0 had not been uninstalled (I had used: $ sudo apt-get remove opencpn) before building the latest.

I was eventually able to get it to work by disabling the br24 plugin in both versions of OpenCPN then re-enabling it in the latest.

This was all on NAVIGATRIX(Ubuntu) which came with OpenCPN 3.? and was upgraded with the 4.0 package, then most recently to 4.1.427 from source.

It's working now, I'm not sure if anything can be gained from examining what happened ?

canboat, 2015-05-25:
Ah yes that explains it I have been caught out by multiple installs as well. Let me check if I can add a detection mechanism.

canboat, 2015-05-27:
I will close this then and open a long term issue for detecting multiple instances of the plugin.

NAHANNIV, 2015-05-27:
Ok with me.

closed 2015-11-27 by canboat