keesverruijt/BR24radar_pi #70: Plugin does not correct for chart skewing

canboat, 2015-05-22:
Some people have skewed charts. I have never seen or encountered such a chart but apparently they really exist, and with charts that are skewed 15 degrees or more.

The skew rotation (?) is not considered when creating the overlay.

canboat, 2015-05-22:
There is a "skew" variable in the viewpoint object that is passed to the plugin. Maybe we should just attempt to use this, and see where that leads.

dav777, 2015-05-22:
Thank you canboat for creating this issue.  I just signed up here and will be happy to test any code.  The waters I sail only have a skewed chart that is zoomed in.  I guess it is because the island naturally sits about 15 degrees off of an East West line.  One of the charts is 12352_3.KAP
   Let me know if you need any more info.

canboat, 2015-05-22:
@dav777 please test the current build. If it still does not work try the following: stop O, edit SkewFactor in the .ini file, start O. The default is 1, try changing it to -1 first.

Please report whether it works, and if not, what skew factor you needed, and what skew factor is reported in the log file at VerboseLog = 4.

dav777, 2015-05-22:
Hi Kees,  I'd like to test it.  This is my first time with github, and I'm still learning Linux/Ubuntu, though long ago a programmer.  The way I installed O was to download from here:http://opencpn-navico-radar-plugin.github.io/
  which I believe is an older version.  How would I download the current build?  Note: I don't have a Linux compilation set up configured.
Thanks,   David

canboat, 2015-05-22:
David,

OK, we'll publish a beta on http://opencpn-navico-radar-plugin.github.io/

dav777, 2015-05-22:
Much appreciated!

canboat, 2015-05-23:
New beta files up at http://opencpn-navico-radar-plugin.github.io with the above 'skewing' fix. 

I downloaded your NOAA chart of Long Island, and it is indeed a buggery chart. Never seen those before!

dav777, 2015-05-23:
Installed v1.2 and enabled it in O.  For some reason the radar was not found.  Just prior, it still worked with v1.1 but with the skewed return issue.  I also tried turning on my radar with the helm plotter and it worked OK.  With the radar on from the plotter, O could still not see it.  Can't think of anything that changed on my end.  Could not stay long at the boat for testing.
Was there other radar connection / ethernet code that changed in v1.2?

canboat, 2015-05-24:
No networking code changed between 1.1 and 1.2, I think.

It helps if you could use multiple OS or at least multiple systems when testing, to rule out platform issues.

dav777, 2015-05-28:
After removing the Navco plotters from contention with radar IP resolution, I have some details which will likely give devs some insight.  First, my settings were always "North Up" and the checkbox "Show shewed raster charts as north-up" was not checked. 1.2 did not fix the problem, wish I had taken a screenshot, but it might have doubled the rotation (which would be a simple sign inversion), but I'm not sure whether the problem increased.  What DID fix the problem was to check the  "Show shewed raster charts as north-up" box.  Did not reinstall 1.1 to confirm the fix worked there as well.  Could perform additional tests such as course up and so forth, please advise if you'd like anything tested.  Thanks again.

canboat, 2015-05-28:
@dav777 have you read my instructions from may 22:

> @dav777 please test the current build. If it still does not work try the following: stop O, edit SkewFactor in the .ini file, start O. The default is 1, try changing it to -1 first.

dav777, 2015-05-29:
Sorry, Wifi has been spotty at the boat, so I didn't have the info with me as I had assumed.  My mistake.  Will confirm this weekend.

dav777, 2015-06-03:
Took the time to test each variation and took screen shots to verify.  It seems that SkewFactor has no effect whether 1 or -1.  The skewed charts are wrong, rotated by what seems to be the same deviation from East West that the skewed chart has, unless the "View Skewed Charts as North Up" box is checked.  When that box is checked, then it is correct.

canboat, 2017-08-08:
@dav777 is this still an issue?

dav777, 2017-08-08:
Sorry, I don't know because I haven't downloaded the recent version.  The boat PC is not running, I'll switching do a different PC due to a power issue.