keesverruijt/BR24radar_pi #91: Multi-color and Multi-color 2 almost no discernable difference.

NAHANNIV, 2015-10-26:
Prefrences->Overlay Display Options

In most cases, "Multi-color" and "Multi-colour 2" settings do not make a discernible difference to the display.

douwefokkema, 2015-11-09:
There is a difference in the threshold level on which a target is shown.
With "Multi-color" the threshold is 50 for the blue color, 100 for green, 200 for red.
For "Multi-colour 2" it is 20 for blue, 100 for green and 250 for red.
Personally I favor the "Multi-colour 2" setting, as it gives a better differentiation in targets. However in cases where weak echos should not raise an alarm  "Multi-color" may be the preferred option. The alarm threshold is always the display threshold level, what you see is what raises the alarm.
The threshold level in the .ini file is not functional in the rel 1.3
I would not consider your point as an issue. But you may of course suggest different threshold levels for the various colors.

NAHANNIV, 2015-11-10:
I did see that the threshold levels are different.
Perhaps the targets I was looking at were all too well defined.
I guess my suggestion would be to read the threshold values from the .ini file so the user can change them without re-compiling;  or even better allow a configurable user color threshold setting GUI.

closed 2016-03-11 by NAHANNIV