keesverruijt/BR24radar_pi #94: a little question

JO0st, 2015-11-26:
Hey guys,
for a university project I am going to be working with a simrad 4G. I have found the paper about the protocol of the BR24, but it mentioned that the authors weren't sure wether it would work with the newer version. I noticed that you are supporting the 4G and are working from the same paper. Since I don't have access to the radar to do tests yet I would like to know if the protocol is any different, and if so what the differences are.
Could you give me some information?

canboat, 2015-11-27:
There are subtle differences in the headers and especially the state reports, but the data is still there on the same multi cast address and with the same resolution (2048 spokes of 512 ranges every revolution).

If you can wait a little I'm refactoring the code rather heavily so after this is done you can read it much more easily -- watch the radarwindow branch.

JO0st, 2015-12-01:
That's good news. Is the sample data available in your repository coming from the 4g or from an earlier version?

canboat, 2015-12-01:
> That's good news. Is the sample data available in your repository coming from the 4g or from an earlier version?

4G

closed 2015-12-21 by canboat