keesverruijt/BR24radar_pi #95: Can the RADAR windows be made dockable...

NAHANNIV, 2015-12-02:
Repeated from Flyspray:

I thought it might be a good time to mention this while the code is being changed.

It would be nice to have the control and standalone window dockable in OpenCPN. 

Quoted from BDBCAT:
"The "right" way to allow docking of any PlugIn is to modify the PlugIn so that it's windows become wxAUI clients, like Dashboard does. When that is done, the control windows and/or dialogs can float as they do now, or dock if desired.

This is a relatively simple modification, using Dashboard as a sample instance to study.

Perhaps the best thing to move this forward is a Feature Request against the Radar PlugIns...."

canboat, 2015-12-02:
Cool, I will try this.

canboat, 2015-12-03:
First attempt is in the radarwindow branch.

Hakansv, 2015-12-03:
A test to build radarwindow in VS2013
Attached file includes 35 errors for your info.
Håkan
[Build_errors_radarwindow.txt](https://github.com/canboat/BR24radar_pi/files/50706/Build_errors_radarwindow.txt)

canboat, 2015-12-03:
https://github.com/canboat/BR24radar_pi/commit/0a164d339d7986c920678c1b62f5a68876b2125b fixes build on win32.

Note that this all very much unfinished, but all testing is welcome.

NAHANNIV, 2015-12-04:
Can the RADAR control windows be dock-able as well ?
I wonder if there is a way to tie the control for the standalone RADAR window to that window?

I could not get data from TCPReplay to display, spoke count was shown in the info window, but none displayed. Are you testing with a live RADAR ?

Great to see this plugin progressing so quickly.

Let us know when there is something you would like tested.

Cheers,
JM.

canboat, 2015-12-04:
> Can the RADAR control windows be dock-able as well ?

That is my thinking yes. The guard zone dialog and alarm windows are not going to be dockable, as they are visible only partly.

But maybe the controls should also have a different look + feel? Not sure yet. 
> I wonder if there is a way to tie the control for the standalone RADAR window to that window.
> 
> 

I was wondering the same. If there are two radar windows and two radar controls I’d like to have them co-displayed. Maybe I will test with a radar control that is in the same panel, so that you always have them side by side. The other experiment will be where we hide the radar controls most of the time and click on the radar to show them.

> I could not get data from TCPReplay to display, spoke count was shown in the info window, but none displayed. Are you testing with a live RADAR ?
> 
I am testing with tcpreplay (and the checked in merrimac.pcap.gz). Make sure you have the latest version, have shader disabled (doesn’t work yet) and move the windows around a bit. Do you get to see the radar range circles or at least the black panel?
> Great to see this plugin progressing so quickly.
> 
> 
Thanks!

canboat, 2015-12-04:
Yes, there is some weird interaction between the OpenGL displays.

Make sure you test with trunk version of O, the 4.1.925 version that I use on Windows is much worse than the latest which I use on OS X.

Kees

> Can the RADAR control windows be dock-able as well ?
> I wonder if there is a way to tie the control for the standalone RADAR window to that window?
>
> I could not get data from TCPReplay to display, [...]

Kind regards,

Kees Verruijt

douwefokkema, 2015-12-05:
Hi Kees,

I do get a radar image now, but only in overlay mode (with heading and 
pos). See screen print attached.

Earlier I had the 2 radar windows nicely on the left, but can't get them 
back again and no radar image in these.

With overlay off :
- m_ri->ProcessRadarSpoke(a, b, line->data, RETURNS_PER_LINE, 
range_meters, nowMillis);  is called from br24Receive::ProcessFrame for 
each frame

- but RadarDrawVertex::ProcessRadarSpoke does not seem to be entered. 
(this is the case with overlay on)

But all together I do not yet completely understand the new structure.

Cheers,  Douwe

douwefokkema, 2015-12-05:
Addition on previous mail:
I said:
RadarDrawVertex::ProcessRadarSpoke does not seem to be entered.
But this his is only the case if no position and heading is available.

With heading and position the call does work, but no radar image.

Cheers,  Douwe

canboat, 2015-12-05:
This is caused by a problem in OpenCPN itself, I have sent a pull request to OpenCPN/OpenCPN. For now just build from canboat/OpenCPN.

canboat, 2015-12-05:
Another update.

With the fixes in OpenCPN in, the drawing problems and fonts disappearing are all fixed.

The radar windows are now drawing fine when not docked. When docked they don't update, but if you float them then redock them, or drag them to a different dock location they do update. Weird. 

I've experimented with making the control dialog part of the AUI manager, but that is not so easy for some reason.

NAHANNIV, 2016-05-17:
Implemented in V2

closed 2016-05-17 by NAHANNIV