keesverruijt/BR24radar_pi #103: V2 docking radar window causes Un-Responsive program

NAHANNIV, 2016-05-17:
After docking the RADAR window OpenCPN becomes un responsive and must be forced closed.
Ubuntu 14.04 on ARM.
Using VDR to generate heading and position.
Using tcpreplay to generate RADAR traffic.

I am docking the RADAR window while radar data is being displayed).
Sometimes it hangs immediately, and the docked area is white, sometimes it works for a while but the next action like bringing up the radar control or resizing the window will cause the hang.

Tried running in debugger, but nothing was caught.

canboat, 2016-05-17:
Aargh, this is bad because
- It works fine on Mac and Windows and Ubuntu 14.04 amd64 for me, although I do see some occasional white on Ubuntu but that disappears on the first redraw.
- The Docking code in wxWidgets is abandon-ware since 2006 or so.

I will see what I can do to reproduce.

canboat, 2016-05-17:
I do notice that wxWidgets 3.0.2 is a lot better behaved, but for an easy install of that it would be easier to upgrade your Ubuntu to 16.04, and that isn't available for a lot of ARM systems yet.

NAHANNIV, 2016-05-17:
I built 3.0.2

00:39:55 EDT: wxWidgets version: wxWidgets 3.0.2 Linux 32 bit wxGTK

I will see about upgrading to 16.04, but may not be possible for this board (Nvidia Tegra TK1).

NAHANNIV, 2016-05-17:
The OpenCPN dashboard seems to work OK on the same system ?
What are the differences ?

canboat, 2016-05-17:
Dunno, it works perfectly for me on my AMD box, with out-of-box wxWidget libraries.

Did you also build O with those exact libraries? Can you check that lsof -p <pid-of-opencpn> shows that only a single set of wx libraries is loaded?

canboat, 2016-05-17:
Maybe test without any other plugins enabled.

canboat, 2016-05-17:
Works for me with Dashboard plugin enabled.

NAHANNIV, 2016-05-17:
ubuntu@tegra-ubuntu:~$ lsof -p 3205
COMMAND  PID   USER   FD   TYPE     DEVICE SIZE/OFF    NODE NAME
opencpn 3205 ubuntu  cwd    DIR        8,1     4096  131074 /home/ubuntu
opencpn 3205 ubuntu  rtd    DIR        8,1     4096       2 /
opencpn 3205 ubuntu  txt    REG        8,1 69509587 7077897 /usr/local/bin/opencpn
opencpn 3205 ubuntu  mem    REG        0,9             4189 anon_inode:dmabuf (stat: No such file or directory)
opencpn 3205 ubuntu  mem    REG        8,1   864296 7082792 /usr/lib/arm-linux-gnueabihf/libxml2.so.2.9.1
opencpn 3205 ubuntu  mem    REG        8,1 16270840 2110002 /usr/share/icons/hicolor/icon-theme.cache
opencpn 3205 ubuntu  mem    REG        8,1   120820 7084028 /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_bluetooth.so
opencpn 3205 ubuntu  mem    REG        8,1    34536 6553655 /lib/arm-linux-gnueabihf/libnss_files-2.19.so
opencpn 3205 ubuntu  mem    REG        8,1    30424 6553754 /lib/arm-linux-gnueabihf/libnss_nis-2.19.so
opencpn 3205 ubuntu  mem    REG        8,1  1492416 7084156 /usr/lib/arm-linux-gnueabihf/libvorbisenc.so.2.0.8
opencpn 3205 ubuntu  mem    REG        8,1   133184 7084599 /usr/lib/arm-linux-gnueabihf/libFLAC.so.8.3.0
opencpn 3205 ubuntu  mem    REG        8,1    55248 6553676 /lib/arm-linux-gnueabihf/libnsl-2.19.so
opencpn 3205 ubuntu  mem    REG        8,1  3911436 7077952 /usr/local/lib/opencpn/libchartdldr_pi.so
opencpn 3205 ubuntu  mem    REG        8,1    22404 6553662 /lib/arm-linux-gnueabihf/libnss_compat-2.19.so
opencpn 3205 ubuntu  mem    REG        8,1    13796 7084683 /usr/lib/arm-linux-gnueabihf/libasyncns.so.0.3.1
opencpn 3205 ubuntu  mem    REG        8,1   281284 7086257 /usr/lib/arm-linux-gnueabihf/libsndfile.so.1.0.25
opencpn 3205 ubuntu  mem    REG        8,1    22788 6553721 /lib/arm-linux-gnueabihf/libwrap.so.0.7.6
opencpn 3205 ubuntu  mem    REG        8,1   285428 7084058 /usr/lib/arm-linux-gnueabihf/pulseaudio/libpulsecommon-4.0.so
opencpn 3205 ubuntu  mem    REG        8,1    26256 6553701 /lib/arm-linux-gnueabihf/libjson-c.so.2.0.0
opencpn 3205 ubuntu  mem    REG        8,1   199844 7081989 /usr/lib/arm-linux-gnueabihf/libpulse.so.0.16.2
opencpn 3205 ubuntu  mem    REG        8,1    18120 7084032 /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_pulse.so
opencpn 3205 ubuntu  mem    REG        8,1   335068  529922 /usr/share/fonts/truetype/dejavu/DejaVuSansMono.ttf
opencpn 3205 ubuntu  mem    REG        8,1   149180 7083904 /usr/lib/arm-linux-gnueabihf/libcroco-0.6.so.3.0.1
opencpn 3205 ubuntu  mem    REG        8,1   133848 7084548 /usr/lib/arm-linux-gnueabihf/librsvg-2.so.2.40.2
opencpn 3205 ubuntu  mem    REG        8,1    14008 7086226 /usr/lib/arm-linux-gnueabihf/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-png.so
opencpn 3205 ubuntu  mem    REG        8,1     5572 7086234 /usr/lib/arm-linux-gnueabihf/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so
opencpn 3205 ubuntu  mem    REG        8,1   116240 1970026 /usr/share/mime/mime.cache
opencpn 3205 ubuntu  mem    REG        8,1  2690880 2102355 /usr/share/icons/gnome/icon-theme.cache
opencpn 3205 ubuntu  mem    REG        8,1   954040 2109995 /usr/share/icons/Humanity/icon-theme.cache
opencpn 3205 ubuntu  mem    REG        8,1 11208582 7077950 /usr/local/lib/opencpn/libgrib_pi.so
opencpn 3205 ubuntu  mem    REG        8,1   876918 7088251 /usr/local/lib/opencpn/libvdr_pi.so
opencpn 3205 ubuntu  mem    REG        8,1   842229 7077951 /usr/local/lib/opencpn/libbr24radar_pi.so
opencpn 3205 ubuntu  mem    REG        8,1  2400844 7077949 /usr/local/lib/opencpn/libwmm_pi.so
opencpn 3205 ubuntu  mem    REG        8,1   742380  529920 /usr/share/fonts/truetype/dejavu/DejaVuSans.ttf
opencpn 3205 ubuntu  mem    REG        8,1 10056588 7077948 /usr/local/lib/opencpn/libdashboard_pi.so
opencpn 3205 ubuntu  DEL    REG        0,4          3997714 /SYSV00000000
opencpn 3205 ubuntu  DEL    REG        0,4          2555916 /SYSV00000000
opencpn 3205 ubuntu  mem    REG        8,1    49536 6553744 /lib/arm-linux-gnueabihf/libbz2.so.1.0.4
opencpn 3205 ubuntu  mem    REG        8,1    41616 2107850 /usr/share/icons/Humanity-Dark/icon-theme.cache
opencpn 3205 ubuntu  mem    REG        8,1    71072 2109595 /usr/share/icons/ubuntu-mono-dark/icon-theme.cache
opencpn 3205 ubuntu  mem    REG        8,1   353824  529908 /usr/share/fonts/truetype/ubuntu-font-family/Ubuntu-R.ttf
opencpn 3205 ubuntu  mem    REG        8,1   333616  529912 /usr/share/fonts/truetype/ubuntu-font-family/Ubuntu-B.ttf
opencpn 3205 ubuntu  mem    REG        8,1    71516 7084552 /usr/lib/arm-linux-gnueabihf/tegra/libnvrm.so
opencpn 3205 ubuntu  DEL    REG        0,4          2424842 /SYSV00000000
opencpn 3205 ubuntu  mem    REG        8,1   277264 7083640 /usr/lib/arm-linux-gnueabihf/libibus-1.0.so.5.0.505
opencpn 3205 ubuntu  mem    REG        8,1    67336 7084569 /usr/lib/arm-linux-gnueabihf/tegra/libnvos.so
opencpn 3205 ubuntu  DEL    REG        0,4          2457611 /SYSV00000000
opencpn 3205 ubuntu  mem    REG        8,1    18336 7084627 /usr/lib/arm-linux-gnueabihf/gtk-2.0/2.10.0/immodules/im-ibus.so
opencpn 3205 ubuntu  mem    REG        8,1    35616 2490793 /var/cache/fontconfig/945677eb7aeaf62f1d50efc3fb3ec7d8-le32d8.cache-4
opencpn 3205 ubuntu  mem    REG        8,1    18064 2490774 /var/cache/fontconfig/2cd17615ca594fa2959ae173292e504c-le32d8.cache-4
opencpn 3205 ubuntu  mem    REG        8,1    68768 2490791 /var/cache/fontconfig/04aabc0a78ac019cf9454389977116d2-le32d8.cache-4
[... cut ...]
opencpn 3205 ubuntu 1108u  0000        0,9        0    4189 anon_inode
opencpn 3205 ubuntu 1109u  0000        0,9        0    4189 anon_inode
opencpn 3205 ubuntu 1110u  0000        0,9        0    4189 anon_inode
opencpn 3205 ubuntu 1111u  0000        0,9        0    4189 anon_inode
opencpn 3205 ubuntu 1112u  0000        0,9        0    4189 anon_inode
opencpn 3205 ubuntu 1113u  0000        0,9        0    4189 anon_inode
opencpn 3205 ubuntu 1114u  0000        0,9        0    4189 anon_inode
ubuntu@tegra-ubuntu:~$

NAHANNIV, 2016-05-17:
perhaps I did not properly install wx3.0.2 on this board?
Radar_pi and OpenCPN both built on same system as test.

will check this evening.

canboat, 2016-05-17:
Make sure you compile with the same wxWidgets version as running.

NAHANNIV, 2016-05-17:
Sorry, how ? Or, how would that not happen if compiled on the same machine ?

canboat, 2016-05-17:
Oh easy, if you have the runtime libraries installed from the archives (e.g. standard Ubuntu version) and downloaded + installed the source version separately and compiled against that.

I was triggered by you mentioning 3.0.2 but maybe you didn't know that the standard for Ubuntu 14.04 is 3.0.0-2 (which is different from 3.0.2)

NAHANNIV, 2016-05-18:
wx-config --version shows 3.0.2

NAHANNIV, 2016-05-18:
I can confirm that the docking works on ubuntu 14.04 64bit x86.

canboat, 2016-05-18:
Can you check that you don't have multiple wx libraries installed? Your `lsof` shows that OpenCPN itself uses `/usr/lib/arm-linux-gnueabihf/libwx_gtk2u_html-3.0.so.0.0.0`, is that where your private 3.0.2 libraries live? I'm saying that because that is where I would expect system libraries to be, not user compiled, or did you not specify `/usr/local` during the configuration of the wx libs?

NAHANNIV, 2016-05-19:
My wxWidgets libraries are in /usr/local/lib.
So I did: sudo ldconfig /usr/local/lib
Confirmed that OpenCpn is now using my wxWidgets libraries.
Same crash.
? Sorry, I am probably doing something wrong.

canboat, 2016-05-19:
Did you build OpenCPN using the local wx widgets headers as well?

NAHANNIV, 2016-05-19:
Do I just re-build ? or do I need to set something to point to the proper headers ?
Sorry, this is a bit over my head.

Thanks for your help.

canboat, 2016-05-19:
Why did you build wxwidgets locally in the first place? should not be needed at all. Just install the dev libraries from the Ubuntu repository.

My vote: throw everything away, including all /usr/local, restart from top of linux instructions.

NAHANNIV, 2016-05-19:
they were not available for ARM (at the time).

NAHANNIV, 2016-05-19:
I will Re-format and reload this system. Will not get to it till the weekend.
Things that I don't understand:
- why everything except the docking is working in the RADAR_PI; Seems unlikely to me that this would be the only thing broken by an incompatible library/header.  
- why docking is working for the OpenCPN Dashboard on the same system.

NAHANNIV, 2016-05-26:
I re-loaded this system and the problem persisted. 
This particular system is the only one that has this problem.
Docking works correctly on Raspberry Pi 3 (Debian Jessie).

I think there may be a problem with the drivers for this  particular board (Nvidia Tegra TK1).

dmesg seems to show a problem, Perhaps you can understand what is happening:

`[ 6126.610792] 15-gk20a.0, pid 4533: 
[ 6126.610798]  in use on_eng_pending busy
[ 6126.610810] TOP: 8000000412e67dbc PUT: 0000000412e684f0 GET: 0000000412e67dbc FETCH: 0000e00412e68480
[ 6126.610810] HEADER: 60003914 COUNT: 01110001
[ 6126.610810] SYNCPOINT 00000000 00001501 SEMAPHORE 00000004 14460600 00000436 00000002
[ 6126.610818] 
[ 6126.610845] gk20a gk20a.0: gk20a_fifo_handle_mmu_fault: mmu fault on engine 0, engine subid 0 (gpc), client 15 (prop 0), addr 0x00000004:0x0726e000, type 2 (pte), info 0x00000f82,inst_ptr 0xdb26b000
[ 6126.610845] 
[ 6126.629987] gk20a gk20a.0: gk20a_fifo_set_ctx_mmu_error: channel 15 generated a mmu fault
[ 6126.638198] gk20a gk20a.0: gk20a_set_error_notifier: error notifier set to 31
[ 6126.638198] 
[ 6126.648607] gk20a gk20a.0: fifo_error_isr: channel reset initated from fifo_error_isr
[ 6128.642081] host1x host1x: opencpn: syncpoint id 21 (gk20a.0_15) stuck waiting 964, timeout=2147481647
[ 6128.642095] host1x host1x: id 1 (disp0_a) min 1 max 1
[ 6128.642104] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6128.642112] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6128.642120] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6128.642127] host1x host1x: id 5 (dsi) min 3 max 0
[ 6128.642134] host1x host1x: id 6 (disp1_a) min 36572 max 36572
[ 6128.642141] host1x host1x: id 7 (disp1_b) min 4 max 0
[ 6128.642148] host1x host1x: id 8 (disp1_c) min 4 max 0
[ 6128.642155] host1x host1x: id 9 (gk20a.0_1) min 502027 max 502027
[ 6128.642163] host1x host1x: id 11 (gk20a.0_2) min 270 max 270
[ 6128.642170] host1x host1x: id 12 (gk20a.0_4) min 92990 max 92990
[ 6128.642177] host1x host1x: id 13 (gk20a.0_6) min 81233 max 81233
[ 6128.642184] host1x host1x: id 14 (gk20a.0_7) min 395 max 395
[ 6128.642192] host1x host1x: id 15 (gk20a.0_8) min 968 max 968
[ 6128.642199] host1x host1x: id 16 (gk20a.0_9) min 8740 max 8740
[ 6128.642206] host1x host1x: id 17 (gk20a.0_10) min 11194 max 11194
[ 6128.642213] host1x host1x: id 18 (gk20a.0_11) min 5841 max 5841
[ 6128.642219] host1x host1x: id 19 (gk20a.0_12) min 9762 max 9762
[ 6128.642226] host1x host1x: id 20 (gk20a.0_13) min 2758 max 2758
[ 6128.642233] host1x host1x: id 21 (gk20a.0_15) min 960 max 964
[ 6128.642240] host1x host1x: id 22 () min 3906 max 3906
[ 6128.642247] host1x host1x: id 23 () min 1253 max 1253
[ 6128.642255] host1x host1x: id 26 (vblank0) min 366865 max 0
[ 6128.642262] host1x host1x: id 27 (vblank1) min 324086 max 0
[ 6130.642011] host1x host1x: opencpn: syncpoint id 21 (gk20a.0_15) stuck waiting 964, timeout=2147479647
[ 6130.642026] host1x host1x: id 1 (disp0_a) min 1 max 1
[ 6130.642034] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6130.642042] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6130.642049] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6130.642056] host1x host1x: id 5 (dsi) min 3 max 0
[ 6130.642063] host1x host1x: id 6 (disp1_a) min 36572 max 36572
[ 6130.642071] host1x host1x: id 7 (disp1_b) min 4 max 0
[ 6130.642077] host1x host1x: id 8 (disp1_c) min 4 max 0
[ 6130.642085] host1x host1x: id 9 (gk20a.0_1) min 502027 max 502027
[ 6130.642092] host1x host1x: id 11 (gk20a.0_2) min 270 max 270
[ 6130.642099] host1x host1x: id 12 (gk20a.0_4) min 92990 max 92990
[ 6130.642106] host1x host1x: id 13 (gk20a.0_6) min 81233 max 81233
[ 6130.642113] host1x host1x: id 14 (gk20a.0_7) min 395 max 395
[ 6130.642120] host1x host1x: id 15 (gk20a.0_8) min 968 max 968
[ 6130.642127] host1x host1x: id 16 (gk20a.0_9) min 8740 max 8740
[ 6130.642133] host1x host1x: id 17 (gk20a.0_10) min 11194 max 11194
[ 6130.642140] host1x host1x: id 18 (gk20a.0_11) min 5841 max 5841
[ 6130.642147] host1x host1x: id 19 (gk20a.0_12) min 9762 max 9762
[ 6130.642153] host1x host1x: id 20 (gk20a.0_13) min 2758 max 2758
[ 6130.642160] host1x host1x: id 21 (gk20a.0_15) min 960 max 964
[ 6130.642166] host1x host1x: id 22 () min 3906 max 3906
[ 6130.642173] host1x host1x: id 23 () min 1253 max 1253
[ 6130.642180] host1x host1x: id 26 (vblank0) min 366985 max 0
[ 6130.642187] host1x host1x: id 27 (vblank1) min 324206 max 0
[... 102 lines cut ...]
[ 6140.644380] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6140.644424] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6140.644465] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6140.644502] host1x host1x: id 5 (dsi) min 3 max 0
[ 6140.644540] host1x host1x: id 6 (disp1_a) min 36641 max 36641
[ 6140.644577] host1x host1x: id 7 (disp1_b) min 4 max 0
[ 6140.644612] host1x host1x: id 8 (disp1_c) min 4 max 0
[ 6140.644649] host1x host1x: id 9 (gk20a.0_1) min 502226 max 502226
[ 6140.644689] host1x host1x: id 11 (gk20a.0_2) min 270 max 270
[ 6140.644728] host1x host1x: id 12 (gk20a.0_4) min 93172 max 93172
[ 6140.644765] host1x host1x: id 13 (gk20a.0_6) min 81233 max 81233
[ 6140.644801] host1x host1x: id 14 (gk20a.0_7) min 395 max 395
[ 6140.644836] host1x host1x: id 15 (gk20a.0_8) min 968 max 968
[ 6140.644874] host1x host1x: id 16 (gk20a.0_9) min 8740 max 8740
[ 6140.644911] host1x host1x: id 17 (gk20a.0_10) min 11194 max 11194
[ 6140.644947] host1x host1x: id 18 (gk20a.0_11) min 5841 max 5841
[ 6140.644983] host1x host1x: id 19 (gk20a.0_12) min 9762 max 9762
[ 6140.645019] host1x host1x: id 20 (gk20a.0_13) min 2758 max 2758
[ 6140.645143] host1x host1x: id 21 (gk20a.0_15) min 960 max 964
[ 6140.645182] host1x host1x: id 22 () min 3906 max 3906
[ 6140.645219] host1x host1x: id 23 () min 1253 max 1253
[ 6140.645260] host1x host1x: id 26 (vblank0) min 367584 max 0
[ 6140.645296] host1x host1x: id 27 (vblank1) min 324806 max 0
ubuntu@tegra-ubuntu:/usr/local/bin$ 
`

NAHANNIV, 2016-05-26:
[ 6126.610792] 15-gk20a.0, pid 4533: 
[ 6126.610798]  in use on_eng_pending busy
[ 6126.610810] TOP: 8000000412e67dbc PUT: 0000000412e684f0 GET: 0000000412e67dbc FETCH: 0000e00412e68480
[ 6126.610810] HEADER: 60003914 COUNT: 01110001
[ 6126.610810] SYNCPOINT 00000000 00001501 SEMAPHORE 00000004 14460600 00000436 00000002
[ 6126.610818] 
[ 6126.610845] gk20a gk20a.0: gk20a_fifo_handle_mmu_fault: mmu fault on engine 0, engine subid 0 (gpc), client 15 (prop 0), addr 0x00000004:0x0726e000, type 2 (pte), info 0x00000f82,inst_ptr 0xdb26b000
[ 6126.610845] 
[ 6126.629987] gk20a gk20a.0: gk20a_fifo_set_ctx_mmu_error: channel 15 generated a mmu fault
[ 6126.638198] gk20a gk20a.0: gk20a_set_error_notifier: error notifier set to 31
[ 6126.638198] 
[ 6126.648607] gk20a gk20a.0: fifo_error_isr: channel reset initated from fifo_error_isr
[ 6128.642081] host1x host1x: opencpn: syncpoint id 21 (gk20a.0_15) stuck waiting 964, timeout=2147481647
[ 6128.642095] host1x host1x: id 1 (disp0_a) min 1 max 1
[ 6128.642104] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6128.642112] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6128.642120] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6128.642127] host1x host1x: id 5 (dsi) min 3 max 0
[ 6128.642134] host1x host1x: id 6 (disp1_a) min 36572 max 36572
[ 6128.642141] host1x host1x: id 7 (disp1_b) min 4 max 0
[ 6128.642148] host1x host1x: id 8 (disp1_c) min 4 max 0
[ 6128.642155] host1x host1x: id 9 (gk20a.0_1) min 502027 max 502027
[ 6128.642163] host1x host1x: id 11 (gk20a.0_2) min 270 max 270
[ 6128.642170] host1x host1x: id 12 (gk20a.0_4) min 92990 max 92990
[ 6128.642177] host1x host1x: id 13 (gk20a.0_6) min 81233 max 81233
[ 6128.642184] host1x host1x: id 14 (gk20a.0_7) min 395 max 395
[ 6128.642192] host1x host1x: id 15 (gk20a.0_8) min 968 max 968
[ 6128.642199] host1x host1x: id 16 (gk20a.0_9) min 8740 max 8740
[ 6128.642206] host1x host1x: id 17 (gk20a.0_10) min 11194 max 11194
[ 6128.642213] host1x host1x: id 18 (gk20a.0_11) min 5841 max 5841
[ 6128.642219] host1x host1x: id 19 (gk20a.0_12) min 9762 max 9762
[ 6128.642226] host1x host1x: id 20 (gk20a.0_13) min 2758 max 2758
[ 6128.642233] host1x host1x: id 21 (gk20a.0_15) min 960 max 964
[ 6128.642240] host1x host1x: id 22 () min 3906 max 3906
[ 6128.642247] host1x host1x: id 23 () min 1253 max 1253
[ 6128.642255] host1x host1x: id 26 (vblank0) min 366865 max 0
[ 6128.642262] host1x host1x: id 27 (vblank1) min 324086 max 0
[ 6130.642011] host1x host1x: opencpn: syncpoint id 21 (gk20a.0_15) stuck waiting 964, timeout=2147479647
[ 6130.642026] host1x host1x: id 1 (disp0_a) min 1 max 1
[ 6130.642034] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6130.642042] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6130.642049] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6130.642056] host1x host1x: id 5 (dsi) min 3 max 0
[ 6130.642063] host1x host1x: id 6 (disp1_a) min 36572 max 36572
[ 6130.642071] host1x host1x: id 7 (disp1_b) min 4 max 0
[ 6130.642077] host1x host1x: id 8 (disp1_c) min 4 max 0
[ 6130.642085] host1x host1x: id 9 (gk20a.0_1) min 502027 max 502027
[ 6130.642092] host1x host1x: id 11 (gk20a.0_2) min 270 max 270
[ 6130.642099] host1x host1x: id 12 (gk20a.0_4) min 92990 max 92990
[ 6130.642106] host1x host1x: id 13 (gk20a.0_6) min 81233 max 81233
[ 6130.642113] host1x host1x: id 14 (gk20a.0_7) min 395 max 395
[ 6130.642120] host1x host1x: id 15 (gk20a.0_8) min 968 max 968
[ 6130.642127] host1x host1x: id 16 (gk20a.0_9) min 8740 max 8740
[ 6130.642133] host1x host1x: id 17 (gk20a.0_10) min 11194 max 11194
[ 6130.642140] host1x host1x: id 18 (gk20a.0_11) min 5841 max 5841
[ 6130.642147] host1x host1x: id 19 (gk20a.0_12) min 9762 max 9762
[ 6130.642153] host1x host1x: id 20 (gk20a.0_13) min 2758 max 2758
[ 6130.642160] host1x host1x: id 21 (gk20a.0_15) min 960 max 964
[ 6130.642166] host1x host1x: id 22 () min 3906 max 3906
[ 6130.642173] host1x host1x: id 23 () min 1253 max 1253
[ 6130.642180] host1x host1x: id 26 (vblank0) min 366985 max 0
[ 6130.642187] host1x host1x: id 27 (vblank1) min 324206 max 0
[ 6132.642254] host1x host1x: opencpn: syncpoint id 21 (gk20a.0_15) stuck waiting 964, timeout=2147477647
[ 6132.642323] host1x host1x: id 1 (disp0_a) min 1 max 1
[ 6132.642371] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6132.642415] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6132.642456] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6132.642495] host1x host1x: id 5 (dsi) min 3 max 0
[ 6132.642539] host1x host1x: id 6 (disp1_a) min 36572 max 36572
[ 6132.642579] host1x host1x: id 7 (disp1_b) min 4 max 0
[... 93 lines cut ...]
[ 6140.644333] host1x host1x: id 1 (disp0_a) min 1 max 1
[ 6140.644380] host1x host1x: id 2 (disp0_b) min 1 max 0
[ 6140.644424] host1x host1x: id 3 (disp0_c) min 1 max 0
[ 6140.644465] host1x host1x: id 4 (disp0_d) min 1 max 0
[ 6140.644502] host1x host1x: id 5 (dsi) min 3 max 0
[ 6140.644540] host1x host1x: id 6 (disp1_a) min 36641 max 36641
[ 6140.644577] host1x host1x: id 7 (disp1_b) min 4 max 0
[ 6140.644612] host1x host1x: id 8 (disp1_c) min 4 max 0
[ 6140.644649] host1x host1x: id 9 (gk20a.0_1) min 502226 max 502226
[ 6140.644689] host1x host1x: id 11 (gk20a.0_2) min 270 max 270
[ 6140.644728] host1x host1x: id 12 (gk20a.0_4) min 93172 max 93172
[ 6140.644765] host1x host1x: id 13 (gk20a.0_6) min 81233 max 81233
[ 6140.644801] host1x host1x: id 14 (gk20a.0_7) min 395 max 395
[ 6140.644836] host1x host1x: id 15 (gk20a.0_8) min 968 max 968
[ 6140.644874] host1x host1x: id 16 (gk20a.0_9) min 8740 max 8740
[... cut ...]
[ 6140.644947] host1x host1x: id 18 (gk20a.0_11) min 5841 max 5841
[ 6140.644983] host1x host1x: id 19 (gk20a.0_12) min 9762 max 9762
[ 6140.645019] host1x host1x: id 20 (gk20a.0_13) min 2758 max 2758
[ 6140.645143] host1x host1x: id 21 (gk20a.0_15) min 960 max 964
[ 6140.645182] host1x host1x: id 22 () min 3906 max 3906
[ 6140.645219] host1x host1x: id 23 () min 1253 max 1253
[ 6140.645260] host1x host1x: id 26 (vblank0) min 367584 max 0
[ 6140.645296] host1x host1x: id 27 (vblank1) min 324806 max 0
ubuntu@tegra-ubuntu:/usr/local/bin$

canboat, 2016-05-26:
Does it work when you use the vertex array instead of shaders? OpenGL is a bit of a beast to manage, there is so much that can go wrong and we just can't give a guarantee that it will work.

I'm trying to get my RPi3 to work with the hardware OpenGL driver, but not much luck so far -- black screen...

NAHANNIV, 2016-05-26:
Vertex Array or Shader, Data or No Data, Nmea or No Nmea, makes no difference. On this system sometime after docking the radarwindow OpenCPN gets hung up; Somtimes it is immediate (docked radarwindow is blank) and sometimes it takes up to 20seconds (docked radarwindow works correctly for 20 seconds). I can dock and undock the control window without a problem.

The OpenGL support on Rpi is a work in progress. I was able to install the latest BR24radar_pi on version 7 of OpenPlotter (kernel 4.1) and it seemed to work well, but have not been able to get it to work with the latest Raspbian (kernel 4.4).

NAHANNIV, 2016-05-28:
Commits to 4c088e1d47fccbff0c4b4dde08f4e54473d08949 change the behaviour on my problem system.

Now it eventually seems to recover after a 10-20 second pause.

On my formerly working Ubuntu 14.04 X86 64 system the "NO RADAR" message is no longer working, when there is no RADAR I now get a blank white screen.

canboat, 2016-06-07:
Can we close this issue?

NAHANNIV, 2016-06-07:
sure, never really figured it out, I'll open a new issue if I find any new clues.

closed 2016-06-07 by NAHANNIV

canboat, 2016-06-08:
There’s a “reopen issue” button at the bottom ...