keesverruijt/BR24radar_pi #116: Radar Window reverts to default size when re-opened.

NAHANNIV, 2016-05-22:


NAHANNIV, 2016-05-22:
In docked mode.

canboat, 2016-05-28:
Either I don't have this issue, or it is already fixed after my commits to fix #115 and #118.

Can you recheck and if so tell me exactly what is wrong? I did the following steps:

- Start O,
- Dock 2 radar windows on right side and make them half the screen width,
- Quit O,
- Start O.

The radar windows were still docked in the same way and half the screen width. Then retried with a narrow window, same result.

NAHANNIV, 2016-05-28:
Ubuntu x86 64 
Single RADARWindow
Docked
drag edge to half screen
Close RADAR with toolbar
Open RADAR with toolbar
Window is back to default docked size.

canboat, 2016-05-28:
Duh, you're right. I was testing with quit/start of O, not stop/start of radar.

canboat, 2016-05-29:
Should be fixed now. Thanks!

commit 24935bd1 (2016-05-29): Fix #116. Use awful hack to fix this.

wxAUI doesn't allow direct access to the width/height of a dock area. So I load the perspective into a string, parse + modify the string then restore the new string as a perspective. As I say, rather a hack... but it works...
closed 2016-05-29 by canboat