keesverruijt/BR24radar_pi #118: OpenCPN sometimes crashes when plugin is enabled without OpenGL support

NAHANNIV, 2016-05-26:
Ubuntu 14.04 x86 and ARM

On further investigation just enabling and disabling the plugin (regardless of OpenGL) crashes OpenCPN. Usually on second or third click.

------GDB output--------
(opencpn:25351): Gtk-CRITICAL **: IA__gtk_widget_set_size_request: assertion 'width >= -1' failed
[New Thread 0x7fffc307c700 (LWP 25448)]
[Thread 0x7fffc307c700 (LWP 25448) exited]

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff69d371e in wxAuiManager::Update() ()
   from /usr/local/lib/libwx_gtk2u_aui-3.0.so.0
(gdb)

canboat, 2016-05-26:
Can you add the stack backtrace?

canboat, 2016-05-26:
Does this happen with your RPi?

NAHANNIV, 2016-05-26:
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff69d371e in wxAuiManager::Update() ()
   from /usr/local/lib/libwx_gtk2u_aui-3.0.so.0
(gdb) bt
#0  0x00007ffff69d371e in wxAuiManager::Update() ()
   from /usr/local/lib/libwx_gtk2u_aui-3.0.so.0
#1  0x00007fffc6d1c8e9 in br24::RadarPanel::Create() ()
   from /usr/local/lib/opencpn/libbr24radar_pi.so
#2  0x00007fffc6d13f0f in br24::RadarInfo::Init(int) ()
   from /usr/local/lib/opencpn/libbr24radar_pi.so
#3  0x00007fffc6ce1b6a in br24::br24radar_pi::Init() ()
   from /usr/local/lib/opencpn/libbr24radar_pi.so
#4  0x000000000071ba6a in PlugInManager::UpdatePlugIns (this=0xe64a00)
    at /home/opa/OpenCPN/src/pluginmanager.cpp:550
#5  0x000000000071be21 in PluginPanel::SetEnabled (this=0xc8de00, 
    enabled=<optimized out>) at /home/opa/OpenCPN/src/pluginmanager.cpp:3691
#6  0x00007ffff5d173ee in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const () from /usr/local/lib/libwx_baseu-3.0.so.0
#7  0x00007ffff5e8a4e2 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#8  0x00007ffff5e8a846 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#9  0x00007ffff5e8a8ce in wxEvtHandler::TryHereOnly(wxEvent&) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#10 0x00007ffff5e8a963 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
---Type <return> to continue, or q <return> to quit---
#11 0x00007ffff5e8a9c5 in wxEvtHandler::ProcessEvent(wxEvent&) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#12 0x00007ffff5e8a757 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#13 0x00007ffff64139b1 in wxgtk_button_clicked_callback ()
   from /usr/local/lib/libwx_gtk2u_core-3.0.so.0
#14 0x00007ffff21385e7 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#15 0x00007ffff2151088 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#16 0x00007ffff2151ce2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#17 0x00007ffff26b8585 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#18 0x00007ffff21383b8 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007ffff2149557 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007ffff2151a29 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007ffff2151ce2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#22 0x00007ffff26b74d9 in ?? ()
---Type <return> to continue, or q <return> to quit---
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#23 0x00007ffff2759815 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#24 0x00007ffff21383b8 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#25 0x00007ffff2149afb in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#26 0x00007ffff21516f9 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#27 0x00007ffff2151ce2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#28 0x00007ffff2869684 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#29 0x00007ffff2757fc4 in gtk_propagate_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#30 0x00007ffff275837b in gtk_main_do_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#31 0x00007ffff23d23ec in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#32 0x00007ffff17d0e04 in g_main_context_dispatch ()
[... 36 lines cut ...]
#51 0x00007ffff5d1afb7 in wxAppConsoleBase::ProcessPendingEvents() ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#52 0x00007ffff63af2d2 in wxApp::DoIdle() ()
   from /usr/local/lib/libwx_gtk2u_core-3.0.so.0
#53 0x00007ffff63af3e3 in wxapp_idle_callback ()
   from /usr/local/lib/libwx_gtk2u_core-3.0.so.0
#54 0x00007ffff17d0ce5 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#55 0x00007ffff17d1048 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#56 0x00007ffff17d130a in g_main_loop_run ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#57 0x00007ffff2757447 in gtk_main ()
---Type <return> to continue, or q <return> to quit---
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#58 0x00007ffff63c2185 in wxGUIEventLoop::DoRun() ()
   from /usr/local/lib/libwx_gtk2u_core-3.0.so.0
#59 0x00007ffff5d59ca0 in wxEventLoopBase::Run() ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#60 0x00007ffff5d194cd in wxAppConsoleBase::MainLoop() ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#61 0x00007ffff5da564d in wxEntry(int&, wchar_t**) ()
   from /usr/local/lib/libwx_baseu-3.0.so.0
#62 0x00000000004d6788 in main (argc=1, argv=<optimized out>)
    at /home/opa/OpenCPN/src/chart1.cpp:830
(gdb)

NAHANNIV, 2016-05-26:
At the moment, I am unable to get back to a working version of the Rpi.

The dbg backtrace is from my x86 64 ubuntu 14.04

In OpenCPN go to the toolbar->plugins->BR24radar_pi  and click enable, disable, enable till it crashes.

Is it crashing on windows as well ?

canboat, 2016-05-26:
Haven't tried Windows yet, but it does crash on Mac, so this is at least not confined to your system.

canboat, 2016-05-28:
@NAHANNIV can you check whether this commit fixes the crash for you?

NAHANNIV, 2016-05-28:
Crash seems to be fixed. But can't open Emulator after this fix.

canboat, 2016-05-28:
You were right, the emulator no longer worked. But this could happen with real radar data as well, so good catch! Can you try again?

NAHANNIV, 2016-05-28:
Seems to work.

canboat, 2016-05-28:
In that case feel free to close an issue :-)

commit 77f0d0f6 (2016-05-28): Fixes #118. Disconnect pane from AUI manager on close.
commit dcbc419e (2016-05-28): Fix #118 possible deadlock on exit

Made more obvious when using emulator.
closed 2016-05-28 by canboat