keesverruijt/BR24radar_pi #120: No easy way to toggle RADAR Overlay on/off

NAHANNIV, 2016-05-28:
To set-up radaroverlay, I must bring up radarwindow, then use controlwindow to enable radaroverlay, then close radarwindow.

Now if I want to turn off the radaroverlay, I have to bring up the control window and turn it off.

To turn it back on start again.

Also, there seems to be an extra click of the toolbar Icon required if the radar window is hidden.

I previously suggested seperate toolbar icons for window and overlay, that's the best I can think of, but I'm open to any other ideas to make this easier.

canboat, 2016-05-29:
I'm not so keen on multiple toolbar icons, especially if you enable a few plugins there are plenty already.

I agree that the way the system behaves now is not optimal yet. I'm thinking that we need the equivalent of the "pages" button on the plotter that allows you to choose whether you want to see the radar windows or not.

Let me create another experiment. Coming up later today...

NAHANNIV, 2016-05-29:
I do want both, but for different things.

Overlay is great for validating charts when approaching unfamiliar areas. 

RadarWindow removes the clutter of the chart, and lets me focus on RADAR only objects like non-AIS equipped vessels or weather systems; Although it could benefit from EBL and AIS overlay eventually.

canboat, 2016-05-29:
Wow, that was more work than I thought! I've cleared up some internal state variables so that I make a distinction between whether the radar window is shown and whether the radar should be transmitting. This was mixed up in a bad way.

The toolbar button now acts as follows:

* If you've got overlay running and nothing is shown, open the control dialog for the corresponding radar.
* Else if you have no radar and no radar control on screen open the radar window(s).
* Else hide the radar windows and controls.

Also you can now disable the automatic hide of the control dialogs or set the timeout to 10 or 30s in the options dialog.

So let's say you have nothing on but the last time you ran with overlay only and you have auto-hide on. Press icon, then transmit. Overlay reappears. Control disappears after 10s. To disable transmit, press icon then standby.

Now you want to switch to separate radar windows. Open the control if necessary (press icon) then "Show Radar" (or "Show radars" for 4G with dual mode enabled). Optional click on "overlay" to disable the overlay.

Note that this doesn't test well when you're running with a capture file, as that doesn't listen to the transmit/standby commands.

canboat, 2016-05-29:
One of the reasons I'm doing it this way is to be align the UI as much as possible with the Navico plotters. A next round might be to order buttons the same way, then even mimic the UI?

NAHANNIV, 2016-05-30:
Emulator seems broken when in standby (transmit?), it flashes, this would be confusing.

I will try this out with live RADAR at my boat later this week.

canboat, 2016-05-30:
You can close one of the radars by clicking on the close button on the window itself. Having explicit buttons "open radar A / open radar B" is the alternative, but I'm not so sure that is an improvement.

No worries, a good tester is one that isn't afraid to be critical. Keep up the good work.

Good idea to have emulator track the transmit state; this was lost when we switched from keeping internal state to having the button clicks just issue a radar command (which does nothing in case of emulator) and the button text reflect the actual radar status.

NAHANNIV, 2016-05-30:
Here is an Idea that might simplify things: what if clicking the toolbar Icon were to just enable/disable the RADAR "view".

Enable would bring up the last saved view, or just the control panel if there was no RADAR in "view", and possibly start transmission.

Disable would save the "view", remove overlay and window(s) and possibly turn off transmit.

By "view" I mean: enabled/disabled overlays, enabled/disabled radarwindows A or B or Both with position and size.

The concept of Master and Slave might be a worthwhile addition to the preferences dialog as well. If defined as Master disabling the radar View would also stop transmitting. Or perhaps just restart transmit if it was enabled when the view was saved.

Also, it would be handy to be able to access the preferences dialog from the control menu.

canboat, 2016-05-30:
That's a neat idea actually. I kept thinking in terms of show/hide radar windows, but it could also show/hide overlay. The entire point is that you want to be to understand what a button does. And this way it works like many other buttons (show/hide AIS, for instance).

The only issue is how to do you get to the radar control if you don't have the radar window(s) shown but the overlay is enabled? Should the icon then just bring up the control instead of hiding the overlay?

NAHANNIV, 2016-05-30:
I think the icon should be simple on/off.
When toggling on always show control. Then timeout, or not based on preference.

Show RADAR control was in the context menu at one point.
I don't think it needs to be easy to get to since it will seldom be used with overlay; I find the automatic settings are usually very good.

NAHANNIV, 2016-05-30:
Perhaps you could also default the hiding of the control to never, and once someone is comfortable with the RADAR operation then they can set it to auto-hide.

I think that might be simpler for new users.

canboat, 2016-05-31:
Yes, I think that is better. It will probably be coming weekend before I have time for more coding though.

NAHANNIV, 2016-06-06:
The new toolbar Icon logic seems a lot simpler to me.

I think the Show RADAR Control needs to be in the OpenCPN context menu.

For 4G Dual RADAR, I wish there were a way to have one RADAR as the overlay and the other one in the window, and be able to toggle this configuration on and off from the toolbar. Now it always comes back with two windows even after I close one.

canboat, 2016-06-06:
> I think the Show RADAR Control needs to be in the OpenCPN context menu.

I focused on operation with a tablet, where context menus are a pain, but this can be added. It would only be for the overlay right? 

> For 4G Dual RADAR, I wish there were a way to have one RADAR as the overlay and the other one in the window, and be able to toggle this configuration on and off from the toolbar. Now it always comes back with two windows even after I close one.
I hadn't tested that scenario, but it is indeed a valid one -- let me see how this can be done.

NAHANNIV, 2016-06-06:
Yes, context menu "Show RADAR Control" is needed when only overlay is displayed and Control is timed out.

NAHANNIV, 2016-06-12:
Closing this and opening the two remaining issues separately.

commit aa9065eb (2016-05-29): Fix #120

- Add a 'Show/Hide Radar(s)' button on the control dialog.
- Add a 'Control Menu Auto Hide' selection on the options dialog.
- Simplify the actions that happen when you press the Toolbar icon.
- Save desired transmit and 'show radar' state separately.
commit b40588b6 (2016-05-30): Fix #120 - Improve emulator

Emulator now only transmits when you ask for transmit, just like a real radar.
commit c48e5740 (2016-06-04): Fix #120 - Make toolbar button a simple on/off switch of all radar UI.
closed 2016-06-12 by NAHANNIV