keesverruijt/BR24radar_pi #121: Gaurd Zone Dialog not working

NAHANNIV, 2016-05-30:


NAHANNIV, 2016-06-06:
Ubuntu x86 64 14.04

[image]

canboat, 2016-06-06:
Indeed it's funny looking. Weird that this is so different on GTK.

NAHANNIV, 2016-06-06:
It works OK in the old version on the same system:

[image]

canboat, 2016-06-06:
This is what I get on kubuntu 16.04 x86_64:

[image]

Kinda weird that it should work on one system but not another.

canboat, 2016-06-06:
Can you try it now? I fiddled a little with the disable options and the way the zone type options are drawn. Do the controls for range and bearing go disabled/enabled now depending on the zone type?

NAHANNIV, 2016-06-07:
Still looks the same.

There is only a box for inner range, I can use the TAB key to select Outer, start, and end.
When selected the line below the item turns red.
I can enter data but can't see it.
When circle is selected Start and end are inactive.

canboat, 2016-06-07:
Is the dialog the same height as when you’re not in a Guard Zone?

Can you show a screen dump with radar A in the ‘home’ view of the dialog (with transmit/standby at the top) and radar B with the Guard Zone pane active?

NAHANNIV, 2016-06-07:
The Control window is always the same height.

For some reason I can no longer get the emulator to work with RADAR B.
So here is the opposite of what you asked for:

[image]

canboat, 2016-06-07:
I see what is going on here, the Guard zone dialog is compressed because the radar dialogs are as well. You can see how all buttons are missing the top and bottom pixels.

What resolution is this? 1024x600?

NAHANNIV, 2016-06-07:
1920 * 1080

canboat, 2016-06-07:
Ah, I see the problem now. If I use Ubuntu + Unity and go to Settings > Display and set "Scale for menu and title bars" to 1.5 or higher I get the same issue. I need to scale the height of the button bars with the height of the text, at least on GTK and maybe OS X and Windows.

NAHANNIV, 2016-06-07:
Working with Scale set to 1:

[image]

canboat, 2016-06-07:
Much nicer :-) but even then the buttons are not high enough in Unity. GGGGGGGrrrrr!

Will have to figure out something dependent on OS I guess. And/or maybe even an override in the ini file so that we get a workaround if I get it wrong…

NAHANNIV, 2016-06-08:
All except timed transmit is visible now.

However, the control window seems very big and some buttons seem to have more whitespace top and bottom than others ? For instance, Transmit, Hide Radars, Overlay On Head Up have very little wasted white space, while most of the others do.

[image]

canboat, 2016-06-11:
Still not right on GTK, MSW and MAC work OK.

NAHANNIV, 2016-06-12:
Seems OK on Ubuntu 14.04.

commit 4a91f06a (2016-06-06): Fix #121 - Center options on GTK. Disable options when off.
commit 539d5934 (2016-06-07): Fix #121 - Make button size dependent on font height.
commit 534e53f7 (2016-06-08): Fix #121. Second attempt.

Works correctly on OS X, Windows 7 and GTK with various scale factors.
commit e4602079 (2016-06-11): Fix #121 - Resize main control when shown for first time
closed 2016-06-12 by NAHANNIV