keesverruijt/BR24radar_pi #123: Info Window displayed while editing preferences.

NAHANNIV, 2016-06-06:
If the RADAR is transmitting with overlay on while the preferences menu is displayed, the Info window pops up (incorrectly)showing no position data:
 
[image]

canboat, 2016-06-06:
I will have to think about this. It's not our preferences dialog but the OpenCPN Options dialog that causes this. They don't refresh the main chart window nor anything else when this is open, so we don't get any calls from OpenCPN at all, and thus the plugin complains that there is no boat position, heading or variation.

canboat, 2016-06-11:
Fixed this as much as reasonably possible, see comment on commit 183a9f8.

closed 2016-06-11 by canboat

canboat, 2016-06-11:
I found a different nasty solution that should make this really nicely behaved. 

When opening the message box I check that there is no modal dialog visible. The only modal dialog in O happens to be the Options dialog, so this works well. Even if there are other modal dialogs then the logic is still the same as you don't want pop-ups when a modal dialog is shown.

commit 183a9f8a (2016-06-11): Fix #123, fix #125

This fixes the issues mentioned as much as I can reasonably do in a normal timeframe. Investigation shows that wxOSX is 'more broken' than wxGTK and wxMSW. It just behaves differently -- in particular it ignores the "wxFRAME_FLOAT_ON_PARENT" instruction. Which is why I had wxSTAY_ON_TOP there. Fortunately, FLOAT_ON_PARENT works pretty well on GTK and MSW.

The new behavior is:

- STAY_ON_TOP isn't really STAY_ON_TOP on Mac, if you select a different window then these do draw over OpenCPN. So that is OK.
- On MSW the radar message box drawn during the modal "Options" dialog  [... cut]