keesverruijt/BR24radar_pi #137: RADAR is left on.

NAHANNIV, 2016-06-15:
When the toolbar icon is clicked the RADAR is not turned off,
Icon turns yellow ?

RADAR is not turned off when OpenCPN is closed.

canboat, 2016-06-16:
> When the toolbar icon is clicked the RADAR is not turned off,
> Icon turns yellow ?
> 
Yes, this is on purpose. The yellow indicates that no radar is active on screen.

> RADAR is not turned off when OpenCPN is closed.
> 
> 

It will turn itself off after a while if no display sends “I’m still here”. We don’t turn it off because there might be a different display (chart plotter, other OpenCPN instance on different computer) still open.

NAHANNIV, 2016-06-16:
How long will it stay on for when the plugin is not active ?

This is a big change from the last version. I think there should be some indication that the Scanner is still transmitting; Is that possible ?

My personal preference would be to turn it off, but perhaps that action could be configurable.

canboat, 2016-06-16:
> How long will it stay on for when the plugin is not active ?
> 
I will check.
> This is a big change from the last version.
> 
It is a behavior change that you suggested :-)
Actually all older versions did this when you closed Opencpn without hitting "standby" before closing this as well, so it is not new in that way. It is new that the close/open doesn't stop the scanner, only the "standby" button does.

I'm thinking that maybe I can check in the reader thread whether there is a different device online as well, if that is possible we can make the behavior dependent on this. What I want to prevent is that a 'hide radar' in O also stops the radar for the plotter/other  device. I will check this this weekend.
> I think there should be some indication that the Scanner is still transmitting; Is that possible ?
> 
Yeah, I'm thinking that the green light should stay on but the rest of the button should change appearance when the radar is not shown. Like the AIS and vector label icons.
> My personal preference would be to turn it off, but perhaps that action could be configurable.
> 
Yes, because you only have one device. Let's see what I can come up with. Preferences are a cheap way out, it should behave nicely for everybody depending on the setup.

marcus-aa, 2016-06-16:
Hi, thanks for all the hard work.

Please find a good way that does NOT hard-shutdown the scanner. Last year before Douwe modified the code it was very "rude" and shut down the radar even when the plotter was using it. It was not only inconvenient but quite dangerous.

Just send the "standby" or soft-off command to the radar when OCPN does not use it any more. The scanner will handle everything from there, at least the 4G.

- If no other unit is using it it shuts itself down in a very short time, 10-20 seconds if I remember correctly
- If another unit has requested it, scanner will now increase scanning speed for this unit as it is no longer busy doing 2 things.

If there must be a hard OFF (I see no need for it, but my NSE plotter has it), then put this in the settings, possibly with a warning dialogue "This will turn off radar for all devices"

Under no circumstance should the menu radar button or any other easily accessible button send a hard-OFF command to the radar, but it would be nice if the button icon could indicate if the radar is already/still running even if radar data reception  is not active in OCPN.

BR /Marcus

canboat, 2016-06-17:
I just checked and the actual radar keeps transmitting for 30s after receiving the last "keepalive" packet. This means that it goes into standby automatically after 30s.

My proposal is to stop sending "keepalive" when the radar display is hidden. I sometimes like to remove the overlay for a few seconds, to verify what I am looking at, then open the overlay again. If you keep the radar window hidden for longer than 30s it will go into standby automatically.

I will put redesigned icons that show more clearly what is going on on the TODO list.

PS Marcus, " If another unit has requested it, scanner will now increase scanning speed for this unit as it is no longer busy doing 2 things." is not exactly right. The UDP data can and is received by all units. What is true is that if you use both radar A and radar B that "fast scan" is automatically disregarded. So if you have 1 display using radar A and the other radar B then you see what you describe when you close one of those: the radar starts turning faster. Note that fast scan also only works on certain ranges; I think 2NM and less but am not sure about that.

NAHANNIV, 2016-06-17:
I am OK with 30seconds till standby, but if the RADAR was transmitting when I clicked the toolbar icon to turn off the display, I would like it to be transmitting when I turn the display back on, regardless of how long it has been off. Is that possible?

canboat, 2016-06-17:
Yes. That’s how i made it now. If you click after 30s it will take 1-3 s for the radar to respond though. Do it before it has gone to standby and it is instantaneous (nice no?)

canboat, 2016-06-17:
I'm good with how it is now, it also does what Markus wants: NO implicit STANDBY request is sent UNLESS you press STANDBY (... duh ...) provided we implement #143.

Can you agree?

NAHANNIV, 2016-06-17:
Sounds good to me, I will be able to test at boat on Sunday.

commit 0945b338 (2016-06-17): Fix #137 - Don't send keepalive when hidden.

The radar color "bulb" on the radar icon now reflects whether the radar is transmitting.
I still need to change the icon to show whether the radar is shown or hidden (as well as the tooltip.)
closed 2016-06-25 by canboat