keesverruijt/BR24radar_pi #138: Feature request: scroll wheel or pinch to change range in radar window

NAHANNIV, 2016-06-15:
Using menu for range is annoying with range setting requiring many mouse clicks from the top level menu.

canboat, 2016-06-17:
Super idea!

canboat, 2016-06-18:
Let me know whether this works good enough. I'm seeing a lot of mousewheel events so to prevent too fast zooming I implemented a crude limit of 1 zoom level per second. This is maybe too slow.

marcus-aa, 2016-06-19:
Very good idea, Please make the direction (in/out) configurable.

Wen using a professional navigation joystick that uses a twisting motion for mouse scroll, OCPN zooms the wrong direction....

Often the sensitivity also needs to be configurable to work well with different devices, and it's nice to get faster zoom when you turn further, but a step limiter is also good, try 2-3 steps/sec.

I'll be back at the boat for more testing next week!

canboat, 2016-06-19:
Marcus, good suggestions. Will do.

canboat, 2016-06-20:
Unfortunately "pinch" is only supported in wxWidgets 3.1.0, and OpenCPN runs with 3.0.2 so this will have to wait a little. Double finger scroll becomes zoom on a Mac trackpad, which is fine.

canboat, 2016-06-20:
I'm thinking I could also add "-" and "+" buttons on screen, like the plotters.

marcus-aa, 2016-06-20:
On 2016-06-20, at 21:34, Kees Verruijt wrote:

> I'm thinking I could also add "-" and "+" buttons on screen, like the plotters.

That would be great when running on traditional single-touch monitors.

/M

canboat, 2016-07-13:
Works great now (I could not test with actual radar when committing 8 days ago.)

commit 0ec33b5f (2016-06-18): Fix #138 - Use mousewheel zoom

Works with Mac double finger up/down as well.

The algorithm is a little crude right now: one range change per second allowed.

This also reworks the entire range calculation and interoperation with the plotter, which had turned into a big mess.
commit b5b95c49 (2016-06-20): Fix #138 - Zoom 3 levels per second. Implement zoom direction option in preferences.
commit 99039b1a (2016-06-22): Fix #138 - Increase sensitivity so that a single mouse wheel click produces a range change.
commit ead0e091 (2016-07-04): Fix #138 - Add - and + buttons at bottom
closed 2016-07-13 by canboat