keesverruijt/BR24radar_pi #139: Ebl vrm Cursor placement not working

NAHANNIV, 2016-06-16:
The cursor is placed on he opposite side of the ship from where I click?

NAHANNIV, 2016-06-16:
Happens in North up display with HDT other than 0 or 180.

canboat, 2016-06-16:
Is this when you click in the radar window or when you click in the chart window?

Does your radar have heading input? You can verify this by setting VerboseLog to an odd value (it’s a bit flag nowadays, with 1 = Verbose) and checking for:

        LOG_VERBOSE(wxT("BR24radar_pi: %s transmits heading, using that as best source of heading"), m_ri->name.c_str());

I think we should make that a LOG_INFO so that it is visible even at VerboseLog 0.

NAHANNIV, 2016-06-16:
My autopilot supplies heading data on the boat, I will check the exact sentences and the log when I am there next.

I can reproduce this using VDR and these sentences:

$AIHDT,090.0,T
$GPGGA,174708.000,4311.8317,N,07934.5014,W,1,07,1.5,82.4,M,-34.7,M,,0000
$GPGSA,M,3,26,15,29,13,21,05,18,,,,,,3.6,1.5,3.3
$GPRMC,174708.000,A,4311.8317,N,07934.5014,W,0.00,000.00,151214,,,A

NAHANNIV, 2016-06-16:
Happens with window in North Up when heading != (180 || 0)
clicking on chart or window.

NAHANNIV, 2016-06-16:
Do you mean heading data to the scanner?
I do have the ri-10 box hooked up to the scanner and my NMEA2000 network.
I have never been sure how to test that connection, I asked Douwe when he first implemented it.
Perhaps something could be put in the info window?

My Raymarine 3g autopilot is connected to a seatalk to NMEA2000 converter. Which seems to supply heading at 1hz. I see it translated back to NMEA0183 USB by my VesperXB8000. On my todo list is to try to get 10hz heading from the autopilot.

I do not see the message you asked about in the log.

canboat, 2016-06-16:
> Do you mean heading data to the scanner?
> I do have the ri-10 box hooked up to the scanner and my NMEA2000 network.
> I have never been sure how to test that connection, I asked Douwe when he first implemented it.
> Perhaps something could be put in the info window?
> 
Actually I think you already can, if it says RADAR next to heading instead of HDM or HDT. But this can be made more explicit.
> My Raymarine 3g autopilot is connected to a seatalk to NMEA2000 converter. Which seems to supply heading at 1hz. I see it translated back to NMEA0183 USB by my VesperXB8000. On my todo list is to try to get 10hz heading from the autopilot.
> 
If you have only 1 hz heading my suspicion is that the RI10 will not inject the heading in the radar spokes, and that would possibly explain the behavior you are seeing.

I will see if I can verify this weekend by disconnecting the nmea2000 network from my RI10.
> I do not see the message you asked about in the log.
> 
That would be consistent with my suspicions about it ignoring 1Hz data.

canboat, 2016-06-17:
It should be ok now, can you verify it works for you as well?

NAHANNIV, 2016-06-21:
With the latest, I seem to have a problem with placing the cursor in Head Up mode. Same thing that was previously happening with north up.

Placing the cursor from the chart window seems to work correctly.

NAHANNIV, 2016-06-26:
Bump, this is still a problem with the latest pull.

canboat, 2016-06-26:
Yes, still working on this. Family interfered :-)

Met vriendelijke groeten,
Kees

canboat, 2016-06-26:
Hope to finish this before the sun is down in 2 hrs.

canboat, 2016-06-26:
Guys, please test.

NAHANNIV, 2016-06-27:
looks good.

commit 5a2c0850 (2016-06-17): Fix #139 - Fix EBL/VRM location
commit 2c5cbb8e (2016-06-26): Fix #139 - A little more work than I thought!
closed 2016-06-27 by NAHANNIV