keesverruijt/BR24radar_pi #143: Improve icon in toolbar

canboat, 2016-06-17:
- Use the new "scaled" vector icon display method from O 4.2.
- Draw an X through the icon if the radar is hidden.
- Change the tooltip according to the hidden/show state.

canboat, 2016-07-13:
There are now five different icons:

- Grey on grey = Radar hidden
- Grey on grey with large "?" mark = Radar not seen
- Amber on black = Radar on standby in timed transmit mode
- Green on black = Radar on standby
- Green + Red on black = Radar transmitting

closed 2016-07-13 by canboat

Hakansv, 2016-07-13:
Fix #143 - Increase fatness of radar rings
The initial grey are fine. The rest are still very black, see picture. (And the screen dump is better than live.)
[image]

canboat, 2016-07-13:
Can you measure how large your icons are in pixels?

Hakansv, 2016-07-14:
Measured with a tool (http://www.arulerforwindows.com/) it's 30 x 30
[image]

Hakansv, 2016-07-14:
And BTW my screen resolution is one of the most common, 1366 x 768

Hakansv, 2016-07-14:
New info:
My icons are now slightly better. Strange but I've done two things; PC restart and installed BR24 via the package install.exe instead of just copying the dll. Can't say how either of could made any difference?
But - I still think the black part is somewhat overtaking the picture. What about dark grey??
[image]

commit 6482ab78 (2016-07-13): Fix #143
commit 2aa6eaf8 (2016-07-13): Fix #143 - Increase fatness of radar rings