keesverruijt/BR24radar_pi #148: b&g radar setting for, harbor etc.

rgleason, 2016-06-20:
I have not used the plugin. Hope to.
Noticed three settings for radar in zeus12T which adjust sensitivity for harbor conditions, seaclutter and rain.  Could the plugin do this? Maybe it does already if so, please close.

canboat, 2016-06-21:
Clutter (sea and rain) is already supported (in the "Adjust ..." menu in the latest 'radarwindow' version)

Harbour/Offshore not yet, if you could analyse what commands and data reports are involved would be great.

rgleason, 2016-06-22:
Will do so & advise

canboat, 2016-06-26:
I will probably lift this to v2.x not 2.0.

rgleason, 2017-09-06:
Merrimac thank you for your response on CF. I hope the PCAP file from Eric will help. I understand it is just frosting on the cake and the plugin has all the gain, seaclutter, etc, controls accessible.

Halo has Auto  - Custom, Harbor, Offshore, Weather, Bird Modes.
Zeus Touch 12 has Harbor, See, Offshore modes.