keesverruijt/BR24radar_pi #149: Guard zones (1,2) are not alarming

Hakansv, 2016-06-21:
Radar type: BR24(3G), OS: Win10, OCPN: 4.4.0, RW: 2.614. All built by VS2013. 
Guard zones (1,2) are not alarming, neither sector nor circle.
See attached picture where an alarm should have been received.

[image]

canboat, 2016-06-21:
Weird. I must admit I've been testing this mostly with the separate PPI radarwindow. I'll check that it works with radar overlay only.

canboat, 2016-06-23:
If only all bugs were this easy! Can you close if it works for you as well?

canboat, 2016-06-25:
Sorry, it is still not correct. Values are calculated now but not the correct ones.

commit 6f592a5c (2016-06-23): Fix #149 - One if too many!
commit 3e1f7934 (2016-06-25): Fix #149 - Rework how we compensate for OpenGL rotation
closed 2016-06-25 by canboat