keesverruijt/BR24radar_pi #150: Hide "Window" also close the radar control

Hakansv, 2016-06-21:
Radar type: BR24(3G), OS: Win10, OCPN: 4.4.0, RW: 2.614. All built by VS2013.

Showing radar overlay and window:
When "Hide Window" is selected the Radar Control is also hidden and the only way to get it back (right click menu "Radar control" is grayed) is to:
1 - mouse right click: "Show radar" (Although radar overlay is already shown) and then
2 - "Radar control" is now active and can be used.

canboat, 2016-06-21:
I've seen that the internal state of what is shown is indeed somewhat buggy right now. The control should be left visible if it is the control for an overlay. If it is only controlling the PPI radarwindow I think it is fair that it is closed (as there is no radar that you want to see...)

Hakansv, 2016-06-21:
re: "as there is no radar that you want to see...)"
Well, there is. The chart overlay radar was still working!

canboat, 2016-06-21:
Yes, as I said, that is the bug -- if it is also controlling the overlay it shouldn't close it. But if it isn't (if it is NOT the overlay radar or overlay is off) then it should...

canboat, 2016-06-22:
It's much better now but as the commit says not perfect yet. When you hide the entire radar including overlay but the control window is shown then when you show the overlay again the control should also come back, but that takes a second click (either click the toolbar icon again or right mouse -> Show Control.)

Also the "Show control" should be greyed out if you have only 1 radar enabled and the single control is on screen, or 2 radars enabled and both controls are on screen.

Hakansv, 2016-06-22:
Nicely fixed, thanks.

closed 2016-06-22 by Hakansv