keesverruijt/BR24radar_pi #151: Targets trail is not working

Hakansv, 2016-06-21:
Radar type: BR24(3G), OS: Win10, OCPN: 4.4.0, RW: 2.614. All built by VS2013.

The idea of target trail is very good and welcome but I can't get it working. I've tried all time settings.
Could this log message be connected?:
12:30:35: Error: BR24radar_pi: Radar unhandled control setting for control 18
12:30:36: Error: BR24radar_pi: Radar unhandled control setting for control 18
12:30:40: Error: BR24radar_pi: Radar unhandled control setting for control 18

canboat, 2016-06-21:
It only works on the separate PPI radarwindow. This is exactly like how the plotters work. Besides, it would ruin your chart window with too many echoes.

Hakansv, 2016-06-21:
Ahaa... But if you could get it working on the chart overlay I think I'll use it a lot. When close to a lot other thinks it's easy to turn of if cluttered.

canboat, 2016-06-21:
Brrr, each his own I guess. But I can add it if you really want :-)

Does it work for you on the Position Plot Indicator (= the seperate radar window?)

Hakansv, 2016-06-21:
Didn't test it actually. Can do tomorrow. (Not to be ungrateful but the radar window is not my first priority. I love the charts overlay)

canboat, 2016-06-21:
I hear you. I thought it would be pretty useless myself, the PPI plot, but I must say that with the EBL/VRM and trails on it is very useful.

Hakansv, 2016-06-21:
Maybe looking for impossibilities but if trails was only affecting moving targets and not land masses it will normally not clutter that much? (Apart from outside Rotterdam or so....)

marcus-aa, 2016-06-21:
On 2016-06-21, at 23:03, H�kan Svensson wrote:

> Maybe looking for impossibilities but if trails was only affecting moving targets and not land masses it will normally not clutter that much? (Apart from outside Rotterdam or so....)

Now we are talking about V4.0 or something like that...

There is a Swedish company called SeaCross that has made a fantastic radar/chart system. They call it "radar underlay" or NVT and I have had a chance to use it both in real world and during METS 2009 where we shared a booth.

It is by far the best chart/radar combination I have seen to date, including ECDIS-approved systems on commercial ships...

Instead of putting the radar data on top of the chart, they take relevant parts of the chart, correlate with the radar echoes, and displays that "under" the PPI image, or a fusion of PPI and chart. It paints known targets (land contours, bouys, etc, in a different colour from unknown echoes (like other ships).

http://www.seacross.se/software-rad/

http://www.seacross.se/wp/wp-content/uploads/2012/09/l_radar_4.png

/Marcus

canboat, 2016-06-21:
Cool. But I think we should probably start stop adding features and close the remaining bugs. With PPI display with EBL/VRM and trails I think this a really worthy 2.0 version.

canboat, 2016-06-22:
It's enabled on the overlay now, but the transparency still needs work (it is not transparent enough). But I ran out of time for this morning.

canboat, 2016-06-22:
Note: It's only on in multi-color mode (duh! It is multicolor!) which is now reflected in the control dialog as well. Whether to show it on the overlay is set in the preferences. Default is OFF.

Hakansv, 2016-06-22:
Nice! I'll ride to boat now for a test

Hakansv, 2016-06-22:
Now testing (git pull 15 min ago)
I've trials in radar window but not in overlay. Multi color is on BUT the checked option is not persistent in preferences and is not shown in radar control although the overlay has multi colors.
There is neither no new option in preferences!

Hakansv, 2016-06-22:
I also tried to set "Radar1Trails = 3" in the ini file without effect.

canboat, 2016-06-22:
Aww sorry I forgot to “git push”, done now.

Hakansv, 2016-06-22:
Much better!
Nice trials also in charts overlay. Very handy for weak occasionally targets, thanks. And, yes, new options are present in preferences but no multi color info in radar control, if intended.

closed 2016-06-22 by Hakansv

canboat, 2016-06-22:
If you have preferences set to Mono color then Target Trails should be disabled.

> On 22 Jun 2016, at 11:59, Håkan Svensson <notifications@github.com> wrote:
> 
> Much better!
> Nice trials also in charts overlay. Very handy for weak occasionally targets, thanks. And, yes, new options are present in preferences but no multi color info in radar control, if intended
> 
> 
Yes functionally it is much like a visual “multi sweep filter” for the guard zones.

But I predict that you won’t like it so much when the boat is moving. For that we need a “true motion” display, but that will be v3.0 :-)