keesverruijt/BR24radar_pi #155: Timed transmit issues in branch RadarWindow

Hakansv, 2016-06-23:
On Win10
1)
There is no message in charts overlay to inform user when in standby caused by timed transmit.
 (The former message box wasn't very nice, I know, but at least informative.)

2)
Timed transmit is not switched off, e.g. set to zero, when I force the radar to transmit in standby phase.
It should also be switched off if a guard zone is alarming. Haven't tested due to the not functional alarm.

canboat, 2016-06-23:
> On 23 Jun 2016, at 10:23, Håkan Svensson <notifications@github.com> wrote:
> 
> On Win10
> 1)
> There is no message in charts overlay to inform user when in standby caused by timed transmit.
> (The former message box wasn't very nice, I know, but at least informative.)
> 
> 
I think the best place would be a different icon, but since I want to re-do the icons so they are SVG based (new functionality in O 4.2) let’s wait for that first. SVG is pretty cool, I can dynamically create a file change the color and then create an icon.
> 2)
> Timed transmit is not switched off, e.g. set to zero, when I force the radar to transmit in standby phase.
> It should also be switched off if a guard zone is alarming. Haven't tested due to the not functional alarm.
> 
OK, I was not sure how that should work.

canboat, 2016-06-23:
BTW Håkan, there is a REALLY nice place for the standby/transmit text on the PPI window :-)

But TT can use some more love and care, I will look into it. Maybe keep the TT button on the control dialog visible if you have TT > 0.

commit e74dc8fe (2016-06-25): Fix #155 - Put timeout on standby/transmit button

Reset timed transfer when guard zone is active or user presses standby/transmit.
commit cf71f458 (2016-06-25): Fix #155 - Same text on PPI display as button
closed 2016-06-25 by canboat