keesverruijt/BR24radar_pi #157: Guard Zone Dialog steals control window

NAHANNIV, 2016-06-26:
This makes it difficult to do anything with the RADAR while there is a target in the gaurd zone, including turning off the gaurd zone.

Possible soloutinons:
- different window.
- option to turn off gaurd zone from the alarm window.

Hakansv, 2016-06-26:
The second option sounds good. 
I'm not convinced of the first. When used to this new behavior it's rather handy even though my first impression was like yours.

BTW: Kees, pls recover saving radar menus last position. That function was recently lost.

canboat, 2016-06-26:
> The second option sounds good. 
> I'm not convinced of the first. When used to this new behavior it's rather handy even though my first impression was like yours.
> 
Yeah, I really don't like multiple windows. Especially when they decide they want to be"under" the chart window.
> BTW: Kees, pls recover saving radar menus last position. That function was recently lost
> 
Please do not merge issues, this makes it hard to close them.

canboat, 2016-06-26:
New issue opened for radar menu

canboat, 2016-06-26:
On the original issue, I don't understand. If you confirm the alarm, it doesn't raise again and I can continue using the menu.
[image]

Please give me more detailed steps for what your exact issue is if pressing Confirm doesn't fix it for you.

NAHANNIV, 2016-06-26:
Testing at my slip today with many boats entering/exiting the chanel. After confirming the alarm sometimes the alarm would be raised repeatedly while I tried to navigate back to the gaurd zone menu.

canboat, 2016-06-26:
The “confirmation” resets when the guard zone becomes empty:

  if (bogeys_found) {
    if (!m_guard_bogey_confirmed && TIMED_OUT(now, m_alarm_sound_timeout)) {
      // If the last time is 10 seconds ago we ping a sound, unless the user
      // confirmed
      m_alarm_sound_timeout = now + ALARM_TIMEOUT;

      if (!m_settings.alert_audio_file.IsEmpty()) {
        PlugInPlaySound(m_settings.alert_audio_file);
      } else {
        wxBell();
      }
    }
  } else {
    m_guard_bogey_confirmed = false;  // Reset for next time we see bogeys
  }

so if you have an ultra small guard zone and the boat leaves then the confirmation needs to be made again. Maybe there should be a timeout on this so that a confirmation still holds for a number of seconds after you confirmed.

Hakansv, 2016-06-26:
Ok - I'm convinced. 
Either let the first "confirm" shut all alarms for a while 
or add your second option; a button to set all alarms off 
or move the alarm box out of radar menu

NAHANNIV, 2016-06-26:
Yes the zone was quite small with lots of traffic.
Perhaps this is not a valid test? But it was annoying!

canboat, 2016-06-26:
Fixed it I think. No traffic in sight here right now, and I need to go to bed.

NAHANNIV, 2016-06-27:
Sorry, I have no way to simulate this and won't see the live situation till next weekend.

marcus-aa, 2016-06-27:
Here is my crazy idea; (sorry to bring this up in a bugfix issue, but it is related to the problem)

Don't make alarms that produce anything that needs to be confirmed or reset to keep normal operation of a navigation system. Ever. For any reason.
Many professional manufacturers have managed to make their alarms unusable and dangerous this way, just look at Raymarines AIS alarms on the E-series. Useless.

An alarm should be visible but not intrusive. There is no way to know that this alarm is the most important thing for the mariner in that exact time. 

If I remember correctly, disruptive alarms are not allowed in the ECDIS standard, and for a good reason.

If we see this as purely a "I  am on a crossing without a boat in sight and need to take a nap" kind of alarm, OK, I can turn it off and never use it. But that would be a pity because it could be very useful in other situations as well.

Ideally, there could be 3 levels to select from

1. Off
2. Normal
3. Audible
4. Intrusive

Normal alarm should be visual only, for example the target becomes red and/or flashes, and/or guard zone flashes. Corresponding to the ECDIS standard for dangerous AIS targets.
Audible should be like normal but with a single sound when this state occurs
Intrusive should be like we discuss now, something you cant get to shut up without confirming....

BR
/Marcus

canboat, 2016-06-27:
I don’t think it is crazy at all, in fact I fully agree. Dunno whether we can make this for 2.0.0.

Can you find out what the ECDIS standard for this is precisely?

> Don't make alarms that produce anything that needs to be confirmed or reset to keep normal operation of a navigation system. Ever. For any reason.
> Many professional manufacturers have managed to [...]

marcus-aa, 2016-06-27:
On 2016-06-27, at 12:12, Kees Verruijt wrote:

> Can you find out what the ECDIS standard for this is precisely?

Will try to find relevant parts tonight. I mainly studied the AIS parts earlier.

canboat, 2016-06-27:
I put your suggestion in a new enhancement request, see issue #165.

NAHANNIV, 2016-07-17:
Trying to test this today, but there seems to be a problem: as I am setting it up (still entering the parameters) the guard zone alarm comes up for a second then off for a second, then repeats. 

Once setup, some geese swimming in the harbor provided a moving variable target that also caused the alarm to be raised too often.

canboat, 2016-07-22:
I've been working on this today, see today's commits. 

It works to my satisfaction now.

There is a new option on the options/preferences dialog that allows you to set the repetition of how soon the alarm goes off again. Setting it to 0 will disable audible alarm.

Please reopen this issue if you still get the fast repetition of the guard zone alarm coming up. It should be fixed now.

commit f36a4772 (2016-06-26): Fix #157

When a guard zone becomes empty it needs to be empty for 10s before the confirmation is considered cancelled.
In other words, you don't get triggered when a guard zone is empty for just a few seconds.
commit 486ca6f9 (2016-07-22): Fix #157. Also introduces option setting of how soon an alarm is repeated
closed 2016-07-22 by canboat