keesverruijt/BR24radar_pi #158: Radar control menu location is not saved

canboat, 2016-06-26:
According to Håkan in issue #157 

BTW: Kees, pls recover saving radar menus last position. That function was recently lost.

canboat, 2016-06-26:
The designed behavior is that if you move a PPI radar window that the next time that the radar window comes up directly over where you press the new "Menu" button on the PPI display. Previously you could click anywhere in the PPI display, but this is not usually reserved for setting the cursor.

When you're not moving a PPI window or don't have them displayed the design is that the control menu comes up in the previous location. You're right, it doesn't.

canboat, 2016-06-26:
Can you please test? I like the current behavior. You should not notice anything with the radar windows closed. If you do have one or more open, and you move the panel and close the control then the next time it opens on the top right of the panel (directly over the menu button.)

Hakansv, 2016-06-27:
Position is now saved, thanks.
re: "next time it opens on the top right of the panel (directly over the menu button."
Yes correct positioned but, in Win10, behind the radar window thus invisible. I've to move radar window to see it.

canboat, 2016-06-27:
Getting over/under right in all OS is a crime against humanity in wxWidgets. They’re supposed to solve this for you, but ...

commit 42bf332f (2016-06-26): Fix #158 - Restore control location.

Only set control dialog to menu location if panel was moved this session.
commit 7014f875 (2016-06-27): Fix #158 - Next attempt
closed 2016-06-27 by canboat