keesverruijt/BR24radar_pi #163: v2.628 Guard zone alert comes only once

Hakansv, 2016-06-27:
Win10
After a alert is confirmed by the "confirm" button no alert is ever given for that sector again. 
Tested with emulator standby/transmit as a method to clean the zone. (Or close O)

Hakansv, 2016-06-28:
Now tested live and I can't say the alert function is easy to test. It's not obvious what's going on when there are targets in guard zone(s). Kees, I think you may test it for while yourself but here are some of my observations:
- The "next alert" count down is sometimes frozen.
- The alert threshold setting is unclear. It seems to be 20 but I've seen zone 1 = 7 in the alert box (Is ThresholdMultiSweep=20 also the alert set??) 
- If I'm using radar control some levels down no alert box is shown but only the alert sound.
- I think I would like to have the former alert box back instead of interfere it with radar control and then the opportunity to quite the sound and adjust guard settings while observing the result. 
- This is now more a discussion then an issue. Close if you like.

Hakansv, 2016-06-30:
More live tests and observations:
- Alert info box is  not updating after an alarm. 
When an alert is active the dots count is not updated and as I mentioned before also the countdown is also sometimes frozen. Also can the times left value initially be negative.
- If the alert is no longer valid I would like the alert box to close for example when next alarm countdown is zero.

marcus-aa, 2016-06-30:
> � Alert info box is not updating after an alarm. When an alert is active the dots count is not updated and as I mentioned before also the countdown is also sometimes frozen. Also can the times left [...]

What is the point of the countdown ? 

I only see the need for 4 states;

Guard OFF
Guard ACTIVE
Guard ALARM
Guard ALARM MUTED

/Marcus

Hakansv, 2016-06-30:
It's a countdown, 10 sec, to next sound alert if not confirmed. It's better then a disturbing constant play but very good if I am not fully alert. (E.g. napping :) ) (Comparable to the ECDIS dead man alert sound)

marcus-aa, 2016-06-30:
On 2016-06-30, at 12:55, H�kan Svensson wrote:

> It's a countdown, 10 sec, to next sound alert if not confirmed. It's better then a disturbing constant play but very good if I am not fully alert. (E.g. napping :) ) (Comparable to the ECDIS dead man alert sound)

I understand that the alarm sounds every x seconds if not confirmed (can we change interval ?), but what benefit is there in seeing the number count down? If you are sleeping, you don't see it. If you are awake, you don't need it... :-)

/M

Hakansv, 2016-06-30:
Don't no really. Has been there for a long time. A check that everything is working, maybe?

canboat, 2016-07-02:
I think I polished the behavior enough to close this bug. Feel free to re-open if you think otherwise.

commit 3fba56fa (2016-07-02): Fix #163, Fix #167

Work on GuardZone and feedback.

The guard zone display is only shown if the 'top level' menu is shown, so if you are halfway through editing some control it does not interfere.
The guard zone display is hidden and the top level menu shown again if the bogeys have gone away, even if you have not confirmed.

The confirmation lasts a while, so if a bogey leaves the guard zone but then re-enters within 10 seconds you are not alarmed.

The confirm status is shown in the PPI radar window bottom left, where it now says (confirmed) after a bogey count if the confirmation status st [... cut]
closed 2016-07-02 by canboat