keesverruijt/BR24radar_pi #164: No radar control in overlay mode

douwefokkema, 2016-06-27:
In overlay mode (radarwindows closed) there seems to be no way to get the radar control back. OCPN context menu radarcontrol does nothing. Toolbar button gives no radarcontrol either.
Windows 10, I3 processor, running radar on Tx.

canboat, 2016-06-27:
Is this with the latest pull? I just added a “raise” that should bring it to the top, as Hokan said it was behind the O or PPI window.

If not, can you upload the logfile at VerboseLog=3 ?

douwefokkema, 2016-06-27:
This seems to be the latest commit 5428d80dee370f.

The radar is completely locked now. No way to get the menu (dimmed in the context menu), no way to get radarwindows or switch radar on. Restart of O does not help.
[opencpn.log.txt](https://github.com/canboat/BR24radar_pi/files/334589/opencpn.log.txt)
[opencpn.ini.txt](https://github.com/canboat/BR24radar_pi/files/334591/opencpn.ini.txt)

Hakansv, 2016-06-27:
Douwe 
Test to edit the ini file and set the position values to 10. As of Issue #162 (closed but not yet fixed) is the default control position -1000/-1000 and thus invisible. The ini file post for this is new so first time you use this beta you'll get the default values:
Radar0ControlPosX=-10000
Radar0ControlPosY=-10000
.....
Radar1ControlPosX=-10000
Radar1ControlPosY=-10000

canboat, 2016-06-27:
Hm, I'm seeing weird values (32767 for both X and Y) for the control. Can you remove Radar*ControlPosX and Y from the ini file and try again? Or even remove or rename the BR24radar section from the ini file if that doesn't work.

My bet is that wx on Windows doesn't actually use int internally.

canboat, 2016-06-27:
Hakan, I use -10000 to detect that the position has never been set. Try it -- remove them or reset them to -10000 and the control should just come up in a default position.

Hakansv, 2016-06-27:
Yes now solved and correct! The mismatch occurred when the previous version has been used, as for Douwe and me. Wouldn't disturb others from now on I think.

canboat, 2016-06-27:
Note that the -10000 only is in the log file if you haven't shown that radar control. If you do have shown it at least once it should contain the coordinates the control is at. Oh I see it works for Håkan now, his comment just showed up!

canboat, 2016-06-27:
I still don't get why Douwe has/had +32767 in his position ini file. It may have been something with an intermediate version, indeed. It should no longer be possible to get that value in your ini file (unless you have a 32K screen, of course :-) )

canboat, 2016-06-27:
BTW, I also implemented upgrade from V1 to V2 init files. See bug #136.

douwefokkema, 2016-06-27:
Corrected positions of control in ini file. OK now.

canboat, 2016-06-27:
Super.

closed 2016-06-27 by canboat

douwefokkema, 2016-06-30:
Problem is back again, after playing with the radarwindows:
Radar0ControlPosX=1770377630
Radar0ControlPosY=67515597

commit 621f521d (2016-07-02): Fix #164 - Bind radar control to be in vicinity of OpenCPN main window