keesverruijt/BR24radar_pi #167: Radar Controls is stealing focus

Hakansv, 2016-06-28:
Windows 10
Radar control seems to aggressive in getting focus. When control is open it takes focus every second or so. That makes it impossible to use keyboard shortcuts in OCPN. If I click on O canvas to get focus for shortcuts, like +/- zoom, Radar Controls takes it back in a second. The main purpose would be to give focus to OCPN as much as possible. When sailing I expect a keyboard action to affect O at once.

canboat, 2016-06-28:
Ah this is a result from my adding an extra 'raise' two days ago because you complained it was hidden. Can you try with this line removed, I can't access the source today but you ought to be able to find it pretty easily.

Hakansv, 2016-06-28:
Correct. Removing Raise() in UpdateDialogShown() solves this issue.
You'll will have to find another method to not hide the control behind radar window when the window's icon is used. 

And when focus is in OCPN it takes two clicks to use a radar control menu. One to activate the control and one to open the menu point.
You don't have a windows system to test in, do you?

canboat, 2016-07-04:
The solution I came up with is neither simple nor pretty, but it is as close as I can get without major work. Basically a floating radar A window can overlap the control for radar B and vice versa, but the control of the raised radar window will always be visible.

canboat, 2016-07-04:
oh, and of course docked radar windows never hide any controls anymore.

Hakansv, 2016-07-04:
I'll do live test tomorrow. Radar control is much better and, yes, radar window is not covering it. But the guard alert is a bit nervous. See attached video. --- No not supported. I'll send a mail.

Hakansv, 2016-07-05:
The control is now visible all time, good.
You still need two clicks to activate a menu point when OCPN has focus.
Guard alert blinking waits for solution. (It seems to alter between radar control, short blink, and alert box.)

Hakansv, 2016-07-13:
Commit: Revert to old way of creating control dialog on MSW
- When OCPN has focus I still have to click two times on a radar control button to activate. On first click radar control gets focus, next click activates the button. Using a control in for example grib_pi it's enough with one click for the same scenario.
- Guard alert is still the same; switching between alert and radar control and is steeling focus from OCPN

canboat, 2016-07-22:
@Hakansv I tested today with Windows 10, and I also need to click twice. Let me see whether this is something that can be fixed in the plugin.

canboat, 2016-07-27:
Email from @Hakansv and @douwefokkema confirms reparenting solves this.

commit 3fba56fa (2016-07-02): Fix #163, Fix #167

Work on GuardZone and feedback.

The guard zone display is only shown if the 'top level' menu is shown, so if you are halfway through editing some control it does not interfere.
The guard zone display is hidden and the top level menu shown again if the bogeys have gone away, even if you have not confirmed.

The confirmation lasts a while, so if a bogey leaves the guard zone but then re-enters within 10 seconds you are not alarmed.

The confirm status is shown in the PPI radar window bottom left, where it now says (confirmed) after a bogey count if the confirmation status st [... cut]
commit 5536ada6 (2016-07-03): Fix #167 - Recreate control dialog every show.

A harsh workaround for Microsoft Windows. If you specify a parent and float-on-parent it doesn't work if the parent isn't shown at the time of the create of the chil
d window. By deleting and creating the dialog the problem is worked around.
commit 27f49c9f (2016-07-03): Fix #167 - Further work on OSX and GTK version.
commit 30f8f22e (2016-07-03): Fix #167 - Fix case where radar windows are hidden or unhidden from the control dialog.

There is a slight flicker of the control dialog as it is recreated, but this is better than the alternative.
closed 2016-07-27 by canboat