keesverruijt/BR24radar_pi #171: Hang on de-init of plugin (windows only)

canboat, 2016-07-13:
Mac + Linux work fine.

Seems to be quite recent, didn't have this issue before.

Hakansv, 2016-07-13:
I've noticed this as well after new versions of BR24. After a couple of OCPN up and downs it's OK.  Something in reading config file for a new version??

canboat, 2016-07-21:
In commit 59757b69db5c0609f96381b642a5f19fc2a4359a I fixed at least one issue. As usual with these things you can't really be sure this is definitely the final cause, but I get no further hangs right now on my virtual machines in Windows 7 and Windows 10.

Hakansv, 2016-07-21:
I've tried to stress it hang on de-init but it works fine close nicely.
This issue may have been connected to a on Win exit hang on OCPN itself?? (briefly mentioned before) See also http://www.cruisersforum.com/forums/f134/ocpn-4-2-1-master-branch-win10-halts-on-exit-168883.html#post2168134

canboat, 2016-07-27:
@douwefokkema reports by email:

Overlay on, but radar on standbye. No radarwindows. On close OCPN hangs.
This bug was introduced in cea512db6 (Fix #167)

douwefokkema, 2016-07-28:
Hang only occurs when radar has not been at Tx at all during OCPN session (but radar stand bye). In this case Tx socket is open and not closed before exit

Hakansv, 2016-07-28:
Bug confirmed! Didn't noticed on board.

Den 2016-07-28 kl. 12:24, skrev Douwe Fokkema:

douwefokkema, 2016-08-02:
Just tested with d78838. No hang anymore in the above case. But hanging in many other cases. Such as:
Radar OFF, start O with pi active, radar hidden, close O: hanging.

douwefokkema, 2016-08-04:
[opencpn.log.txt](https://github.com/canboat/BR24radar_pi/files/401034/opencpn.log.txt)

It seems that Tx sockets are not closed on exit, see attached log.

canboat, 2016-08-08:
@douwefokkema confirms now fixed with latest commit 74a734e.

commit d788383b (2016-07-27): Fix #171 - Hang on deinit -- cannot remove draw object until receive thread is gone.
commit bcf7b609 (2016-08-05): Attempt to fix #171 and #187. Wait for thread to finish.
closed 2016-08-08 by canboat