keesverruijt/BR24radar_pi #173: Place EBL not working correctly

NAHANNIV, 2016-07-16:
The EBL seems to be placed -90deg. from where the cursor is.

Also, the VRM circle is the same colour as the range rings.

I liked it when you were drawing a circle at the intersection of EBL and VRM, Why did you drop that ?

marcus-aa, 2016-07-17:
As per S52 Preslib 6.0 :+1: https://www.iho.int/iho_pubs/standard/S-52/S-52_e6.0_EN.pdf

Orange is the mariner's colour, for notes, chartwork, chart corrections. The scale bar,
north arrow, and mariner's navigation objects such as EBLs and VRMs are also
orange.

[image]

canboat, 2016-07-20:
I think the issue itself is fixed.

Marcus, thanks, maybe we should push for a S52 color scheme in OpenCPN itself as well..

Please reopen this issue if you still have issues with the EBL or VRM location. Note that I did NOT remove the circle (=VRM) not sure where you got that from.

closed 2016-07-20 by canboat

NAHANNIV, 2016-07-20:
The EBL placement issue is fixed.

Just to clarify, on my boat platform (Ubuntu14.04 ARM32) I no longer see a small circle drawn around the intersection of EBL and VRM. I think it was working on this platform before ?

[image]

canboat, 2016-07-20:
The image shows what I expect it to show. I don't remember any extra small circles, neither from a visual nor a coding perspective. 

Met vriendelijke groeten,
Kees

NAHANNIV, 2016-07-20:
ok, sorry, I probably misunderstood what I was seeing.

Cheers,
JM.

marcus-aa, 2016-08-02:
Have to ask, what does "R" in "EBL 71°R" mean? I would understand T or M for True or Magnetic, but what is R ?

canboat, 2016-08-02:
Relative to boat, so 0.0R is directly ahead.

commit 42669781 (2016-07-20): Fix #173. FIx calculation of EBL from mouse ebl.