keesverruijt/BR24radar_pi #174: RADAR B control window position not saved

NAHANNIV, 2016-07-17:
RADAR B control window always reverts to initial position.

canboat, 2016-07-22:
The location of the control window is saved, but it is reset when:

- You move the radar window itself with the control menu closed. The first time you then open the control menu it appears near the "Menu" text on the radar window.
- The OpenCPN window is moved or resized and the control menu is no longer "near" the OpenCPN window. This is to facilitate users that have multiple displays attached and then move O from one screen to the other.

I just tested this again and if I do the following:

- Have O open
- Have floating radar window open and control menu open
- Close O
- Start O

all windows come up in the same location as before.

The following sequence also works for me:

- Have O open
- Have floating radar window open and control menu open
- Hide radar using toolbar icon
- Close O
- Start O
- Show radar using toolbar icon

so I'm not sure what you're seeing.

NAHANNIV, 2016-07-22:
"The location of the control window is saved, but it is reset when:

You move the radar window itself with the control menu closed. The first time you then open the control menu it appears near the "Menu" text on the radar window."

I did not understand the logic, and it is not quite as you state: If the RADAR window is moved with the control window open, the next time the RADAR view is cycled off/on (using the toolbar icon) the control window position changes.

NAHANNIV, 2016-07-25:
ok

closed 2016-07-25 by NAHANNIV