keesverruijt/BR24radar_pi #175: Old scan drawn momentarily when radar is turned on.

NAHANNIV, 2016-07-17:
Today I ran the RADAR at the dock before setting out, then turned it off, leaving OpenCPN running.
Once out on the water in a much clearer area I turned the RADAR back on.
Part of the RADAR picture at my dock was displayed momentarily before being replaced by the correct display.

I can also see this if I switch between emulator and live RADAR.

canboat, 2016-07-21:
@NAHANNIV please close if it works for you. I will be able to report back in a couple of days after sailing and gaining access to WiFi again.

NAHANNIV, 2016-07-21:
Not moving for a few days, so I tested this by switching on the emulator, closing radar, turing off emulator, opening radar.

The emulator image still comes up momentarily.

canboat, 2016-07-21:
The way this was fixed does not fix it for the emulator.

Met vriendelijke groeten,
Kees

commit 0c17744b (2016-07-21): Fix #175

Untested, as it is hard to test when not using a moving boat, but this should clear the image when the data is more than a few seconds old.
closed 2016-07-29 by canboat