keesverruijt/BR24radar_pi #176: RADAR window Position not saved

NAHANNIV, 2016-07-18:
RADAR window position is only saved on exit of OpenCPN if the windows are visible.
If the windows are not visible they go to a default position when next opened.

The size is saved OK.

I am not using docked windows.

I think the window positions and sizes should be saved when the RADAR is turned off, or when OpenCPN is closed.

canboat, 2016-07-22:
I cannot reproduce this. In fact, if I have O come up with the radar windows hidden I see a quick flash of the radar window before it executes the hide, and I see the correct (saved) size for a few milliseconds. When I then "show" the radar window(s) they appear in the desired place and size.

You can investigate this by looking in the opencpn.ini/conf file and look for the AUI data, the saved location should be in there. 

For instance my Windows 10 conf file has:

```
[AUI]
AUIPerspective=layout2|name=ChartCanvas;caption=;state=768;dir=5;layer=0;row=0;pos=0;prop=100000;bestw=2995;besth=910;minw=-1;minh=-1;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=BR24radar_pi-0;caption=Radar;state=2099151;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=699;besth=453;minw=256;minh=256;maxw=-1;maxh=-1;floatx=386;floaty=92;floatw=832;floath=704|name=BR24radar_pi-1;caption=Radar B;state=2099151;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=699;besth=453;minw=256;minh=256;maxw=-1;maxh=-1;floatx=570;floaty=120;floatw=512;floath=512|dock_size(5,0,0)=288|
```

which shows that the floating position for my Radar A (only enabled window) is position 386 x 92 and size 832 x 704.

NAHANNIV, 2016-07-22:
Here is the sequence I am using:

- Position/size RADAR and Control window
- turn off radar view using toolbar icon
- shutdow OpenCPN
- Restart OpenCPN
- turn on radar view using toolbar

Control window is correct, but RADAR window has moved.

canboat, 2016-07-22:
That is what I tested and it works for me. Can you do this:

1. Close O with radar window open. Check content of save location in ini file.
2. Close O with radar window not shown. Check content of save location in ini file.
3. Open and then close O after step 2. Check content of save location in ini file.

Is there a difference in the `floatx` and `floaty` values in the AUIPerspective line?

NAHANNIV, 2016-07-23:
I am not sure of exactly what you might want to see, so please excuse my verbose response.

Close O with radar window open. Check content of save location in ini file.

`[PlugIns/BR24Radar]
ChartOverlay=-1
DisplayOption=0
DrawingMethod=1
EmulatorOn=0
EnableDualRadar=1
GuardZoneDebugInc=0
GuardZoneOnOverlay=1
GuardZoneTimeout=30
GuardZonesRenderStyle=2
GuardZonesThreshold=5
IgnoreRadarHeading=0
MenuAutoHide=0
PassHeadingToOCPN=0
RadarAlertAudioFile=
RadarInterface=169.254.6.20
RangeUnits=0
Refreshrate=5
ReverseZoom=0
RunTimeOnIdle=2
ScanMaxAge=6
Show=1
SkewFactor=1
ThresholdBlue=50
ThresholdGreen=100
ThresholdMultiSweep=20
ThresholdRed=200
TrailsOnOverlay=1
Transparency=5
VerboseLog=0
Radar0Rotation=0
Radar0Transmit=2
Radar0WindowShow=1
Radar0ControlShow=1
Radar0Trails=0
Radar0ControlPosX=715
Radar0ControlPosY=335
Radar0Zone0StartBearing=517
Radar0Zone0EndBearing=540
Radar0Zone0OuterRange=92
Radar0Zone0InnerRange=18
Radar0Zone0Type=0
Radar0Zone0Filter=0
Radar0Zone1StartBearing=0
Radar0Zone1EndBearing=0
Radar0Zone1OuterRange=0
Radar0Zone1InnerRange=0
Radar0Zone1Type=0
Radar0Zone1Filter=0
Radar1Rotation=0
Radar1Transmit=1
Radar1WindowShow=0
Radar1ControlShow=0
Radar1Trails=0
Radar1ControlPosX=1457
Radar1ControlPosY=-32
Radar1Zone0StartBearing=0
Radar1Zone0EndBearing=0
Radar1Zone0OuterRange=0
Radar1Zone0InnerRange=0
Radar1Zone0Type=0
Radar1Zone0Filter=0
Radar1Zone1StartBearing=0
Radar1Zone1EndBearing=0
Radar1Zone1OuterRange=0
Radar1Zone1InnerRange=0
[... 147 lines cut ...]
Radar1ControlPosY=-32
Radar1Zone0StartBearing=0
Radar1Zone0EndBearing=0
Radar1Zone0OuterRange=0
Radar1Zone0InnerRange=0
Radar1Zone0Type=0
Radar1Zone0Filter=0
Radar1Zone1StartBearing=0
Radar1Zone1EndBearing=0
Radar1Zone1OuterRange=0
Radar1Zone1InnerRange=0
Radar1Zone1Type=0
Radar1Zone1Filter=0
[Directories]
S57DataLocation=
InitChartDir=/home/ubuntu/Charts
GPXIODir=
TCDataDir=
WMMDataLocation=/usr/local/share/opencpn/plugins/wmm_pi/data/
[AUI]
AUIPerspective=layout2|name=ChartCanvas;caption=;state=768;dir=5;layer=0;row=0;pos=0;prop=100000;bestw=1920;besth=1034;minw=-1;minh=-1;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=150f14dc-bd80-4959-9597-596a5e81fce5;caption=Dashboard;state=2099148;dir=2;layer=1;row=0;pos=0;prop=100000;bestw=150;besth=566;minw=150;minh=566;maxw=-1;maxh=-1;floatx=1762;floaty=269;floatw=158;floath=590|name=BR24radar_pi-0;caption=Radar;state=2099151;dir=2;layer=0;row=1;pos=0;prop=100000;bestw=960;besth=540;minw=256;minh=256;maxw=-1;maxh=-1;floatx=66;floaty=120;floatw=495;floath=524|name=BR24radar_pi-1;caption=Radar B;state=2099151;dir=2;layer=1;row=0;pos=1;prop=100000;bestw=960;besth=540;minw=256;minh=256;maxw=-1;maxh=-1;floatx=570;floaty=120;floatw=620;floath=512|dock_size(5,0,0)=576|dock_size(2,1,0)=152|`

Is there a difference in the floatx and floaty values in the AUIPerspective line?

Yes, I did not move anything.

canboat, 2016-07-23:
Ah it turns out this is different on Linux compared to OSX and Windows. WxWidgets is supposed to hide such differences from us. Sigh. Maybe it is wx that causes it even. Double sigh...

Anyway I can reproduce on Ubuntu. 

I hope the fix is not too convoluted. With AUI it usually is.

NAHANNIV, 2016-07-25:
OK

commit fbd0a4ff (2016-07-23): Fix #176 - Save floating radar window positions even when radar is hidden
closed 2016-07-25 by NAHANNIV