keesverruijt/BR24radar_pi #177: Radar control/menu text not fit.

Hakansv, 2016-07-20:
The new menu "Transmit/Standby" Don't fit in one row in my W10. See pct.

[image]

canboat, 2016-07-21:
@Hakansv can you verify + close if it works, if not please examine my commit and see what you need to change further.

Hakansv, 2016-07-21:
Works fine in Win 10. 
Thanks

canboat, 2016-08-09:
If you reopen, please tell me why. Even better, fix it :-) or at least provide a screenshot.

Hakansv, 2016-08-09:
I've to reopen this issue. Please see picture:
Vin10 - 1280x800
Translated text is visible but printed over the box borders. Readable but not so nice. 
[image]

canboat, 2016-08-09:
Wow, that's a very long text. 

It's not calculated properly because we don't use the original English text ` _T("side lobe suppression") ` in the place where calculate the maximum width of any button. At the moment this just contains some main English texts and the longest English texts, but they need to contain all of them as any short English text can have a long foreign translation.

Note that there is an (arbitrary) max width of 500 pixels imposed right now to prevent pure runaway with even more verbose language translations.

The fix is to go over the list of button texts again and put them all in the "testBox" calculation in `br24ControlsDialog.cpp`.

marcus-aa, 2016-08-09:
I'm not a big fan of translations as I prefer all my software in english, many navigation terms (especially radar terminology) are very hard to translate. But I also accept that many users absolutely prefers localized versions.

Can I humbly suggest the following shorter and hopefully still intuitive labels/translations, that would help with the space problem?

Standby/Transmit could be STBY/TX as on most radars, and write the full state under.

STBY/TX
TRANSMIT

Håkan, for the swedish translations I suggest the following changes;
(Looks like around 15 characters are available)

Riktningsjust.

Antennhöjd

Interferensjust.

Sidolobsreducer. or Sidolobsundertryckn (still too long...)
("Sidolob" is well accepted terminology, I wouldn't know what "kantavsökning" means)

Main Bang = Huvudpuls ? What is this for, thought there was no main bang in FMCW.

canboat, 2016-08-09:
> On 09 Aug 2016, at 21:43, Marcus AAkesson <notifications@github.com> wrote:
> 
> I'm not a big fan of translations as I prefer all my software in english, many navigation terms (especially radar terminology) are very hard to translate. But I also accept that many users absolutely prefers localized versions.
> 
> Can I humbly suggest the following shorter and hopefully still intuitive labels/translations, that would help with the space problem?
> 
> Standby/Transmit could be STBY/TX as on most radars, and write the full state under.
> 
> STBY/TX
> TRANSMIT
> 
Size-wise these are not a limitation, and to make the plugin a little more user friendly I think we should use as few abbreviations as possible, even if they are in common use and well known by professionals?
> Håkan, for the swedish translations I suggest the following changes;
> (Looks like around 15 characters are available)
> 
> 
As my previous comments say, the code can be changed so that it fits. It just makes sense to keep shorter words sometimes also because there is just less clutter on screen.  Not sure (don’t speak Swedish that well) but your suggestions seem to make sense to me.
> Riktningsjust.
> 
> Antennhöjd
> 
> Interferensjust.
> 
> Sidolobsreducer. or Sidolobsundertryckn (still too long...)
> ("Sidolob" is well accepted terminology, I wouldn't know what "kantavsökning" means)
> 
> Main Bang = Huvudpuls ? What is this for, thought there was no main bang in FMCW.
> 
> 
There is no real big main bang as in pulse radar, but once I started working with trails I found that I did have a small center echo, caused by the ship itself I guess. I elected to call this main bang and allow suppression of this echo.

I guess you may actually like to have own ship trails when trails are enabled and set it to 0 so that you get an own ship’s trail. Maybe we should even artificially create one if desired? Personally I found my own trail very confusing so this is why I introduced this. I propose we all use 2.0 in practice for a while and see what we can come up with to improve it further, and/or decide on different default settings. Nothing beats actual use!

Kees

Hakansv, 2016-08-09:
Marcus. Sidolob is very good, thanks, I'll change. And if you like please have a look in crowdin to help me, it's not easy.
But - - My example was to elucidate what can happen using different languages. See attached pictures. I think Kees has a solution to fit them all.
[image]

canboat, 2016-08-09:
We should probably indicate in Crowdin which texts are used for buttons, and should therefore be kept as brief as possible.

I will go over the Dutch ones this week and shorten them, some are ridiculous (and/or wrong)

canboat, 2016-08-09:
Dutch translations in crowdin now done, unfortunately somebody downvoted "Radar" and replaced it with "BR24 radar" ... can someone please upvote my suggestion "Radar"?

nohal, 2016-08-09:
I have marked your original translation as "approved" now, so hopefully noone is going to "attack" it anymore. Czech updated as well, but I yet have to actually look at what I translated :)

canboat, 2016-08-09:
@nohal thanks!
@Hakansv can you verify that it works now with all translations? I did shorten "Local interference rejection" to "Local interference rej." as it considerably shortens the width of the menu controls in English, so this needs to be uploaded in crowdin as well.

Hakansv, 2016-08-10:
There are no changes to branch "radarwindow". One new commit to master branch but I don't think that one is merged with radarwindow. Typo??

canboat, 2016-08-10:
Yes it is on master, I merged radarwindow back on master for the V2.0 release.

Hakansv, 2016-08-10:
OK Back on master - good.
Radar menu width is now much better adjusted. Thanks.
I'll update the POT-file and load it to Crowdin.

[image]

marcus-aa, 2016-08-10:
> 
> On 9 aug 2016, at 22:53, Kees Verruijt <notifications@github.com> wrote:
> 
> There is no real big main bang as in pulse radar, but once I started working with trails I found that I did have a small center echo, caused by the ship itself I guess. I elected to call this main bang and allow suppression of this echo.

Aha! In that case it's actually more like the opposite of Main Bang, but the result is similar...

Main bang problem in a pulse radar is when you transmit, you need to mute the rx to the transmitted pulse and a zone around your ship becomes deaf. Since there is no pulse or TX/RX switch in the Fmcw radar, you can actually see your own ships echo..

I would call it just that, Own ship trail / echo.

Hakansv, 2016-08-10:
Reopening the radar menu issue:
Now it's no longer possible to change any of the "+"  "+10" "-" "-10"  values.

canboat, 2016-08-10:
@Hakansv please one problem per issue. Open a different one please.

canboat, 2016-08-10:
> On 10 Aug 2016, at 12:35, Marcus AAkesson <notifications@github.com> wrote:
> 
> Main bang problem in a pulse radar is when you transmit, you need to mute the rx to the transmitted pulse and a zone around your ship becomes deaf. Since there is no pulse or TX/RX switch in the Fmcw radar, you can actually see your own ships echo..
> 
> I would call it just that, Own ship trail / echo.

But we’re not sure that all radars see the own ship, so we need an *additional* option Own ship trail (preferably, to me, in a different color, and only on the PPI radar window not the chart window — that already has the normal OpenCPN track). I propose we leave that for v3 as I’d much rather spend time on other stuff (like adding AIS info)

closed 2016-08-10 by canboat

marcus-aa, 2016-08-10:
On 2016-08-10, at 14:42, Kees Verruijt wrote:

> But we�re not sure that all radars see the own ship, so we need an *additional* option Own ship trail (preferably, to me, in a different color, and only on the PPI radar window not the chart window [...]

Two different things, track or trail.

I see very little use for own ship echo and/or trail and would rather suppress that to avoid confusion. As you say, it depends on installation, range, sensitivity and a lot of other factors. 

But own ship track, derived from actual GPS positions is absolutely useful

In S52 color scheme when the background is black (chart mode in dusk or night settings), own track and own ship symbols are light gray.

/M

canboat, 2016-08-10:
> On 10 Aug 2016, at 14:52, Marcus AAkesson <notifications@github.com> wrote:
> 
> On 2016-08-10, at 14:42, Kees Verruijt wrote:
> 
> > > I would call it just that, Own ship trail / echo.
> >
> > But we�re not sure that all radars see the own ship, so we need an *additional* option Own ship trail (preferably, to me, in a different color, and only on the PPI radar window not the chart window � that already has the normal OpenCPN track). I propose we leave that for v3 as I�d much rather spend time on other stuff (like adding AIS info)
> 
> Two different things, track or trail.

Sorry, I meant track.

> I see very little use for own ship echo and/or trail and would rather suppress that to avoid confusion. As you say, it depends on installation, range, sensitivity and a lot of other factors.
> 
> But own ship track, derived from actual GPS positions is absolutely useful

That is what I would like to add later on, own ship track, to the PPI window.

> 
> In S52 color scheme when the background is black (chart mode in dusk or night settings), own track and own ship symbols are light gray.

commit 6b398e00 (2016-07-21): Fix #177 - Add 'Standby / Transmit' text to detection width.