keesverruijt/BR24radar_pi #186: Heading from radar broken ?

marcus-aa, 2016-08-01:
When testing on a friends system about 2 weeks ago, we could not get heading from radar to work any more. 

Test system: Ubuntu, Ocpn 4.4.0, Navico 3G + RI10, Hemisphere Vector Compact GPS compass on N2K.
RadarPi reports version 2.626

Same boat last year was working with 1.x plugins and a 4G radar.

The computer has no other input of heading, only from radar. Connecting an NSE plotter confirms Heading on radar:Yes and Heading on Network:Yes, and the plotter works, also with Marpa.

Ocpn seems to want to fallback to COG (Why was this implemented for radar? Dangerous !) and keeps flipping between 0 degrees and the usual arbitrary COG when boat is moored. Radar image is displayed, but oriented incorrectly.

(Can we block COG for radar with a setting?)

Can anyone test this and confirm something is broken?

douwefokkema, 2016-08-03:
I have no problems with the heading on radar, works as usual. Only issue here is that when heading on radar drops away, the info display is displayed immediately, instead of silently switching over to HDM.
I agree with your remarks on COG, COG is not a heading.

marcus-aa, 2016-08-03:
> Do you have other ways of checking if the radar receives a heading, can you connect a Lowrance display? Is the green led on the RI10 interface box on?

As stated:

"Connecting an NSE plotter confirms Heading on radar:Yes and Heading on Network:Yes, and the plotter works, also with Marpa."

We connected my Simrad NSE and confirmed heading is on radar and on the N2K bus.

The N2K bus is not interfaced to the nav computer, but that should not be necessary. 

Is there a log setting that would help find the problem? 

/ Marcus

douwefokkema, 2016-08-04:
Do you have a source for the variation? Heading on radar will only function when variation is available, radar sends only HDM. When you don't have a source of variation, install the WMM plugin.
What version of the pi are you using? Is it the released 1.33 or the beta 2.x?

marcus-aa, 2016-08-04:
>Do you have a source for the variation?

No, I use ONLY true heading!

>Heading on radar will only function when variation is available, radar sends only HDM. 

Not true, if I feed it true heading it sends true back, as confirmed by the plotter and working last year! 

But I suspected last year that variation was applied in Ocpn even when incoming heading was true....

>When you don't have a source of variation, install the WMM plugin.

Shouldnt need it, but we tried with and without

>What version of the pi are you using? Is it the released 1.32 or the beta 2.x?

As stated, 1.x worked last year, 2.x doesnt.

douwefokkema, 2016-08-04:
Thanks, that is clear. Both version 1.3 and 2.x expect only HDM on radar and therefore require a variation. We should adapt  the software to also accept a HDT on radar. But I do not know how the radar indicates the difference between the two. I am sure they set up a bit somewhere. Are you able to make a wireshark recording of the radar data?

douwefokkema, 2016-08-04:
There is one other possible cause. A new ini file setting was introduced in 2.x, IgnoreRadarHeading. This one should have value IgnoreRadarHeading = 0, which is the default. Please check your opencpn.ini for this. If this is OK, it really should work by providing a variation, though you will then have an error in the heading. I did not find any other differences in the coding of this part between 1.3 and 2.x

canboat, 2016-08-04:
> On 04 Aug 2016, at 10:05, Marcus AAkesson <notifications@github.com> wrote:
> 
> Do you have a source for the variation?
> 
> No, I use ONLY true heading!
> 
> 
Well *some* device on your boat will have a variation defined, otherwise there is no “true heading”.This could be a hardwired offset or a computer that calculates variation based on WMM and the geographical position.  Unless you’re using a GPS compass (that has no magnetic detection mechanism.)

> Heading on radar will only function when variation is available, radar sends only HDM.
> 
> Not true, if I feed it true heading it sends true back, as confirmed by the plotter and working last year!
> 
> But I suspected last year that variation was applied in Ocpn even when incoming heading was true....
> 
> 
That could be true.

One issue for Douwe and myself is that we live an area of the world (Netherlands) where the current variation is very low, about 1 degree, so it is very hard to see whether a reported heading is magnetic or true.

> When you don't have a source of variation, install the WMM plugin.
> 
> Shouldnt need it, but we tried with and without
> 
> What version of the pi are you using? Is it the released 1.32 or the beta 2.x?
> 
> As stated, 1.x worked last year, 2.x doesnt.
> 
> 

And how about 1.32 this year? what worked “last year” is not very telling by itself. So many other things can have changed as well.

canboat, 2016-08-04:
Now back online, with a browser, and I see that this boat does indeed have a GPS compass, so that explains the fact that you don't have magnetic heading available. This is hard for us to emulate as all our compasses have a magnetic source as well.

I verified and the data coming from my radar show that it does need variation applied for the data to be True Heading. Maybe there is more data in the radar packet that tells us whether the heading in the radar packets is true or magnetic.

Even without heading you should be able to get the separate radar window to display. Then once the radar is running it should provide a radar heading source.

The reason that it does not work is that we have no 3G models under test within the core developers, and we think it does not produce radar heading in the same field as the 4G. Last year you tested a 4G and this year a 3G. 

Can you provide us with something like:

    sudo tcpdump -w navico_3g_dump.pcap -i eth0 

whilst the radar is running? Preferably, the entire cycle `standby->transmit->change range->standby` whilst commanding this from a Navico device.

marcus-aa, 2016-08-07:
Just to clarify, this is a friends boat and we are not at the same place anymore, we just did some testing when we met up. There is normally no plotter in his boat, we just brought mine over to see what was going on.

It is correct that the main difference from last year is that we used my 4G on his boat for testing last year, but his 3G this year.

>The reason that it does not work is that we have no 3G models under test within the core developers, and we think it does not produce radar heading in the same field as the 4G. Last year you tested a 4G and this year a 3G.

If this is so, then it would of course explain this behaviour. It would mean that NO 3G radars work with HoR in the plugin which would of course be a big drawback. I can confirm however that the NSE reports "Heading on radar", so it's not missing as a feature from the hardware.

>Maybe there is more data in the radar packet that tells us whether the heading in the radar packets is true or magnetic.

I would think so. I have tested with both HDM and HDT into the 4G and the plotter correctly reports back xxx°T or yyy°M

>Even without heading you should be able to get the separate radar window to display.
 It does.

>Then once the radar is running it should provide a radar heading source.

It doesn't see any heading, just COG.

> Can you provide us with something like:
>sudo tcpdump -w navico_3g_dump.pcap -i eth0 

This is a tcpdump of Ocpn operating this 3G radar. No plotter at hand unfortunately.
https://www.dropbox.com/s/vtwrgftrewo1g7v/navico_3g_dump.pcap?dl=0

I will set up my own test system agin later in the week and can provide pcap files of the 4G operating with both HDM or HDT. I have both a Maretron SSC200 for HDM and a couple of Hemisphere GPS compasses.

Regarding variation;
A GPS compass will also report the current variation using a built-in WMM model, so devices that need magnetic heading can reverse-calculate that. It is also possible to configure some models to provide HDG/HDT/HDM sentences at different rates.

canboat, 2016-08-08:
Removing the requirement that it needs to be a 4G. Hopefully this will now work on BR24 and 3G as well.

I've added a preference "Ignore radar heading" so that if the radar heading turns out to be wrong -- for instance we're misinterpreting True versus Magnetic -- you can switch it off and the plugin will use the data coming in from OpenCPN instead. (We already had this, but you had to set it in the ini file directly.)

Marcus, once you have the pcap files showing the difference between a True and Magnetic compass please upload them to a new issue. I will prepare this for you.

closed 2016-08-08 by canboat

marcus-aa, 2016-08-09:
>Hopefully this will now work on BR24 and 3G as well.

Short initial testing on 3G indicates it works. Radar image moves with compass anyhow, will try to see if alignment is OK or off.

marcus-aa, 2016-08-10:
No major rotation of image on 3G with true heading on radar, info shows heading is received and image is correctly oriented.
Probably still off by extra variation correction, but that is to be expected.

commit 983b6925 (2016-08-08): Fix #186