keesverruijt/BR24radar_pi #187: OCPN crashes when disabling radar_pi plugin

douwefokkema, 2016-08-03:
Radar on Standby, start O. Tools, plugins, select br24radar_pi. Click "disable". OCPN crashes.

canboat, 2016-08-03:
Hm this is probably the same bug as the DeInit on close of O. But hopefully this reproduces for me. 

Are you able to attach a debugger before this action and get a stack trace??

canboat, 2016-08-04:
Nope, this works for me as well.

Are you using the RelWithDbgInfo version or Release? The Debug version is NOT compatible with production OpenCPN releases on Windows.

canboat, 2016-08-04:
Maybe test with a full installation package -- `cmake --build . --config release --target package` ?

canboat, 2016-08-04:
I'm running out of ideas now, by the way. I can't get ANY version to crash anymore, Linux / Mac / MS. Also even `helgrind` on Linux doesn't complain much (except for some innocent variable clashes).

canboat, 2016-08-08:
@douwefokkema confirms fixed.

closed 2016-08-08 by canboat