keesverruijt/BR24radar_pi #196: BR24/3G Radar picture is 180 degr misplaced

Hakansv, 2016-08-09:
Win10 - VS2013
OCPN:  4.5 source fetched today 16-08-09
BR24radar_pi: branch radarwindow source fetched today 16-08-09. (Last commit: "Add extra help files for.....")
>>
Radar echos are 180 degr misplaced. (Guard zones are correct and consequently 180 degr misplaced when radar head is adjusted.)
<<
BR24 ver 1.32 and previous "radarwindow" are correct 
[image]

canboat, 2016-08-09:
What do you see on the PPI radar window in heads up and north up mode?

Hakansv, 2016-08-09:
Oooh - sorry not on board but home again. I can go back and and check if important!
I think, absolutely not sure, it was 180 degr false in heads up.

canboat, 2016-08-09:
> Oooh - sorry not on board but home again. I can go back and and check if important!
> I think, absolutely not sure, it was 180 degr false in heads up.
> 
I think it may be because we’re no longer ignoring Heading coming from the radar on BR24 and 3G models. This was added in issue #186. Before we only used this heading field in the data when it came from a 4G. And you might not have a RI-10/11 hooked up to your BR24, and if so, not a compass hooked up?

Please see what happens if you go to Preferences and set the (also new) “Ignore heading from radar”.

If this fixes it, please upload a `pcap` file made with the radar transmitting (an old one is fine, I know you have those).

Hakansv, 2016-08-09:
I don't use the RI-10/11 it's a direct IP connection to my PC, correct. The heading is coming via OCPN and NMEA from a stand alone compass.

OK - I'll test the “Ignore heading from radar”  and also check the PPI window hopefully later this afternoon.

All my pcap's are still here:  https://dl.dropboxusercontent.com/u/5209364/captures.zip

canboat, 2016-08-09:
I just checked with official released v2.0 code built by the Ubuntu packagers (can’t get more official than that) and it works fine with my 4G, so I am now almost 100% sure its this new “feature” (which may turn out to be broken).

> I think it may be because we’re no longer ignoring Heading coming from the radar on BR24 and 3G models. This was added in issue #186. Before we only used this heading field in the data when it came [...]

marcus-aa, 2016-08-10:
It's probably possible that if the radar sees no valid heading from RI10/11 it sends 0° (or some other undefined value and sets a (yet unknown to us) bit to indicate "no valid heading", but we interpret that as a valid heading ?

canboat, 2016-08-10:
> On 10 Aug 2016, at 15:39, Marcus AAkesson <notifications@github.com> wrote:
> 
> It's probably possible that if the radar sees no valid heading from RI10/11 it sends 0° (or some other undefined value and sets a (yet unknown to us) bit to indicate "no valid heading", but we interpret that as a valid heading
> 
> 
Yes, and it probably sends a different value for “no valid heading” than a 4G. The 4G sends 0xffff. BTW Marcus, is it correct in your 3G dump that the ships heading was South East?

canboat, 2016-08-10:
I just noticed that Matt Paulin has difficulties with this with his 3G as well, and in particular this image is interesting:

[image]

it shows that head up mode is out by about 20 degrees, and his variation happens to be 19.5 degrees. Another reason to fix #194!

canboat, 2016-08-10:
I just realised that the two persons who reported issues with this use no RI-10/11. This seems to confirm that the RI-10/11 repackages the packets with the heading data.

closed 2016-08-10 by canboat

marcus-aa, 2016-08-10:
>I just realised that the two persons who reported issues with this use no RI-10/11. This >seems to confirm that the RI-10/11 repackages the packets with the heading data.

Can't follow you here, the RI doesn't participate in IP traffic at all. Or what do you mean with repackages?
The problems reported are consistent with having no heading _connected_ to the radar. Without RI you don't get heading into radar, but there should be no difference if RI is used but not connected to heading source, or if no RI is used... ?

canboat, 2016-08-10:
> Can't follow you here, the RI doesn't participate in IP traffic at all. Or what do you mean with repackages?
We have no way of knowing how the RI-10 works. I do know it has a bigger SOC than you'd expect. I was expecting something like in a IS40 or ST10, that takes the N2K data and send it in some simple serial format to the radome on one of the free RJ45 wires, but in my memory there is a huge chip in there which suggests it does a bit more. One way for this to make sense is that it actually acts as a router, and takes the IP data coming from the radar and adds the heading to the packets. The other way is that it sends the N2K heading and sends it as Ethernet packets to the radar which then adds that data directly. But this is all just conjecture...

> The problems reported are consistent with having no heading connected to the radar. Without RI you don't get heading into radar, but there should be no difference if RI is used but not connected to heading source, or if no RI is used... ?
Correct.

Hakansv, 2016-08-11:
Now tested on my BR24 radar scanner. Works great also with heading from OCPN. Radar picture position is at the correct angle as well as the guard zones. Both chart overlay and radar window are correct. All concerned properties settings was at default stage.
The issue is solved for me as well.

MattPaulin, 2016-08-12:
Ok, if it's fixed, were can I download a Windows version with the patch? I really must learn how to compile stuff sometime! Thx everyone.

Hakansv, 2016-08-12:
Matt.
Here's mine: https://dl.dropboxusercontent.com/u/5209364/br24radar_pi-2.0-win32.exe
From today's source including commit : "Show firmware date"
Pls also report if radar menu on your system is viewed correct. (Compare issue #197 )

MattPaulin, 2016-08-12:
Thanks Hakan, 
[image]
Radar menu in that version is not useable on my PC...

Hakansv, 2016-08-12:
Kees has changed so now it's fine on my Win10. My link is updated.
https://dl.dropboxusercontent.com/u/5209364/br24radar_pi-2.0-win32.exe
Pls try this though it's compiled using OCPN resent master O version 4.5.0. Hopefully no difference for you and BR24radar_pi. If not please revert and I've a O 4.5.0 install exe

MattPaulin, 2016-08-13:
Thanks Hakan, I'll install on the boat in the morning and report back - off out to dinner right now.

Cheers

Matt Paulin
+64221539176
www.neptunes-gear.com

MattPaulin, 2016-08-13:
Tested that one, same issue, menu unreadable.

Hakansv, 2016-08-13:
Matt. Thanks for your time to test this.
To be sure and to sort out far outs I've tested the new BR24.._pi on O 4.4.0 and that's not the cause, worked fine for me.
I can't say I've any particular next step to test. 
Kees - any ideas? 
....Old remains in the config file??
....How come the big difference between Matt's plugin windows, BR24/Tactics? BR24 has round corners the others sharp? It's not like that on my system, see my screenshot.

[image]

Hakansv, 2016-08-13:
Matt. How's the radar info box?

canboat, 2016-08-13:
> ....Old remains in the config file??
> ....How come the big difference between Matt's plugin windows, BR24/Tactics? BR24 has round corners the others sharp? It's not like that on my system, see my screenshot.
> 
> 
The tactics data is (I guess) not a separate wxDialog but an AUI box, and it is certainly not using wxButton, there are millions of different ways to style windows dialogs anyhow.

Hakansv, 2016-08-13:
Misunderstanding...
What I meant to emphasize was: Why are the boxes so different on Matt's system but equal on mine?

Hakansv, 2016-08-14:
@MattPaulin Please try again. We've made a small change for a test: https://dl.dropboxusercontent.com/u/5209364/br24radar_pi-2.0-win32.exe

MattPaulin, 2016-08-14:
Great, all working fine with this one!! Thanks guys :-)

commit 672a06ee (2016-08-10): Fix #196 and fix #194
commit b57ee3c1 (2016-08-10): Fix #194. Fix #196.

Further fixes for true heading, turned out to have some more corner cases that did indeed also cause
wrong rotation.
commit a9db0fd5 (2016-08-10): Fix #194. Fix #196.

Further fixes for true heading, turned out to have some more corner cases that did indeed also cause
wrong rotation.