keesverruijt/BR24radar_pi #197: Radar control buttons  "+" "+10" "-" "-10" are not longer functional.

Hakansv, 2016-08-10:
After last fix to issue #177 it's no longer possible to change any of the "+" "+10" "-" "-10" values.

Hakansv, 2016-08-10:
Supplementary info:
The increase/decrease buttons for "Installation" menu is not working, they are all zero. 
Other menus buttons works as usual.

Hakansv, 2016-08-10:
My last comment was with an old sv_SE.po file. When now tested with fresh one from a fresh POT file no increase/decrease buttons are working ?? 
I'll see if I can find something. But you may have a hint?

Hakansv, 2016-08-10:
The same problem when using English language. Is it working for you?
(Sorry for my sometimes to quick conclusions.)

canboat, 2016-08-10:
Sorry, I have no issues in English or Dutch. If you want you can send me the swedish POT and I can test with that for you.

Hakansv, 2016-08-11:
On OCPN 4.5.0 and Win10
I've a fresh build from master and using default, English, language.
The radar menu is compressed and no increase/decrease buttons are working. 

[image]

Hakansv, 2016-08-11:
More info:
I've tried recent Canboat master at my on board PC where a live BR24 radar was connected.
All menus works and looks great. Increase/Decrease buttons as expected. Very good.

So what's with my at home build PC?? 
It's OK that some Increase/Decrease buttons won't work using the emulator. 
But it's still an issue with the compressed size as of previous post. I've tried using a new config file created from scratch - no change.
Some graphical mismatch? But it was working before version 2.0! Or is it still a radar not responding effect since I'm on the emulator??  
Any hints for further tests?

Hakansv, 2016-08-11:
Could the weird menu have something to do with this log message?
20:36:52: BR24radar_pi: OpenCPN does not pass OpenGL context. Resize of OpenCPN window may be broken!

canboat, 2016-08-11:
No, that is just a remnant, I worked around it in a different manner and has nothing to do with this.

The height of the button is calculated, and for some reason that does not work for you. We use this code:

  wxStaticText* testButton2Text =
      new wxStaticText(this, ID_BPOS, wxT("Button\ndata"), wxDefaultPosition, wxDefaultSize, wxALIGN_CENTRE | wxST_NO_AUTORESIZE);
  testButton2Text->SetFont(m_pi->m_font);
  testBox->Add(testButton2Text, 0, wxALL, 2);

  ...

#define BUTTON_BORDER 4
#ifdef __WXMAC__
#define BUTTON_HEIGTH_FUDGE 1 + BUTTON_BORDER
#else
#define BUTTON_HEIGTH_FUDGE 1 + 2 * BUTTON_BORDER
#endif

  g_smallButtonSize = wxSize(width, testButtonText->GetSize().y + BUTTON_BORDER);
  g_buttonSize = wxSize(width, testButton2Text->GetSize().y * BUTTON_HEIGTH_FUDGE);

so I suggest you add some debug statements around this area.

Hakansv, 2016-08-12:
Kees.
I've checked which commit caused the changed behavior to my Radar menu.
The commit: "Remove 'requested transmit/receive' and update status using radar report. " is the cause. 
I tried to redo this change in above commit:
@@ -437,7 +437,7 @@ void br24ControlsDialog::CreateControls() {
 
   wxStaticText* testMessage =
       new wxStaticText(this, ID_BPOS, label, wxDefaultPosition, wxDefaultSize, wxALIGN_CENTRE | wxST_NO_AUTORESIZE) Commit:;
 -testMessage->SetFont(m_pi->m_font) Commit:;
 +testMessage->SetFont(m_pi->m_fat_font) ;
   testBox->Add(testMessage, 0, wxALL, 2);

That is:
**br24ControlsDialog.cpp row 440**
//testMessage->SetFont(m_pi->m_fat_font);
  testMessage->SetFont(m_pi->m_font);
When I make this change I get a proper radar menu appearance 

Could you try this change for any inconveniences on your systems?

canboat, 2016-08-12:
Oh, ok.

I changed this because with the “normal” font, some stuff doesn’t fit nicely on the “Value edit” (for instance, “Interference Rejection”, when bold, is a little bigger.)
But I can revert this change and just add 5% to be sure.

Hakansv, 2016-08-12:
OK - very good. I'll make a test when done
Thanks

Hakansv, 2016-08-12:
Win10 successfully tested. Thanks

closed 2016-08-12 by Hakansv