keesverruijt/BR24radar_pi #200: Control menu alternating between standby and Tx

douwefokkema, 2016-08-12:
see https://www.dropbox.com/s/7kqxcormz2fihdi/20160812_111541.mp4?dl=0

[opencpn.log.txt](https://github.com/canboat/BR24radar_pi/files/414975/opencpn.log.txt)
Occurs when replaying a double channel file. Not fully reproducible, occurs sometimes.

It seems that radar state is alternating between transmit and standby, but radar display is not interrupted

canboat, 2016-08-12:
Is this when the logfile is “looped”? I had some issues with that myself (there was a very short “standby” at the start of the file), and I don’t think we should blame the code for that.

douwefokkema, 2016-08-12:
No, that is not the case.

Just found it: radar is in Tx (replay). m_wanted_state = = 
RADAR_STANDBY. So it keeps trying to shut of the radar by calling 
FlipRadarState. FlipRadarState sets m_stayalive_timeout = 0.

Still l do not like the endless repetition of turn_on and off commands. 
If the radar is OK it will react in a single command. I never saw it 
fail in 1.3 (single command). In case it does not react, the user should 
be aware of it that the system is lousy, if he has to press a second 
time too often, he should look after his system.

canboat, 2016-08-12:
I just thought of a different way of getting the radars back to the transmitting state if they were transmitting when O was closed, which is even nicer and faster, and does away with the timers.

commit 053aeb3e (2016-08-13): Fix #200
closed 2016-08-13 by canboat