keesverruijt/BR24radar_pi #201: Trail apperance

Hakansv, 2016-08-13:
Enhancement for further considerations: 
A trail is growing darker, from white to dark gray, during its duration time.
Wouldn't it be more intuitive if it grows the other way, from dark gray to white?
Douwe: 
"I agree, at least for the overlay. In the overlay the contrast between the white part of the trail and the light blue sea is small. But against the dark background of the radarwindows the colors are just fine. Different colors for trails in overlay and radarwindow? Some work again. Or is there an other color that would fit both radarwindow and overlay?"
Kees: 
"The radars do this with increasing transparency, but I found that I could use more different steps than they use, and using different colours makes the code much simpler (drawing code doesn’t need to understand trails.)
Maybe we should just put the colours (start color + end color) in the ini file then you can experiment. And then we can have a public vote on what is wanted.
Even nicer, add a “color picker” in the preferences dialog. Should be pretty simple — there is a wxColourPickerCtrl."

canboat, 2016-08-13:
Colors can now be chosen.

For now there is no possibility to choose different colors for overlay and PPI window. The implementation is a bit wrong here, there is a single RadarDraw object shared between the selected radar and the overlay. To be able to have different colormaps, they should be separate, but this is a bit of a larger overhaul -- I propose we leave that for now.

Hakansv, 2016-08-16:
Works greate

closed 2016-08-16 by Hakansv