keesverruijt/BR24radar_pi #204: With Mono Color display no trails are shown

douwefokkema, 2016-08-13:
Mono color should only effect the primary radar image and not the trails. I expect trails to show better with mono color radar, more simple image.

canboat, 2016-08-13:
The mono color is used by the Shader drawing method to only have a single color, no map at all, so this is hard to do given the current implementation. If we do this there is no performance advantage anymore to single color shader.

Given that choice, do you still think we should do this?

douwefokkema, 2016-08-13:
I just tried it out for the vertex. Removed couple of if (m_pi->m_settings.display_option == 1) and the like, works fine, very good image, less messy. Can we do it only for vertex ?

canboat, 2016-08-14:
> On 13 Aug 2016, at 15:46, Douwe Fokkema <notifications@github.com> wrote:
> 
> I just tried it out for the vertex. Removed couple of if (m_pi->m_settings.display_option == 1) and the like, works fine, very good image, less messy. Can we do it only for vertex ?
> 
> 
Or we just rip out the mono color code for Shader. I don’t think we need the performance gain, really.

canboat, 2016-08-14:
And really we should allow a palette. Since we can do color selection for trails now, why not do this for all colors?

canboat, 2016-08-14:
When we do the palette, we can remove the mono color as you could just select the same color for weak, medium and strong returns.

canboat, 2016-08-17:
Fixed, mono color mode has been removed. You can now just select the same color for weak, intermediate and strong colors.

closed 2016-08-17 by canboat