keesverruijt/BR24radar_pi #207: The problem with displaying text in Cyrillic (Russian localization)

bc38, 2016-08-14:
The problem with inscriptions display in the window of the radar in Cyrillic (Russian) confirmed. Menu button also does not correctly displayed. Screenshot attached. Labels and Menu in ENG displayed normally. He made a full translation plugin on crowdin.
http://www.cruisersforum.com/forums/attachment.php?attachmentid=129436&d=1471022937

canboat, 2016-08-14:
@Hakansv will create a new set of PO files soon from Crowdin.

canboat, 2016-08-14:
It's a bug in TexInfo.cpp, which I borrowed from OpenCPN.

closed 2016-08-14 by canboat

Hakansv, 2016-08-15:
Crowdin has been updated with new strings to translate more or less every second day. Development is still ongoing so I haven't pushed translations files to Canboat source for a while. I'll do that when source is stable and most translators are ready.
Of course - I you wish to test I can push what we have so far, please advice.

canboat, 2016-08-15:
I think we can wait, the word “Menu” now seems to be correct.

commit 0c5db33d (2016-08-14): Fix #207