keesverruijt/BR24radar_pi #210: True vs Magnetic heading confusion

canboat, 2016-08-18:
We have several sources for heading and variation but these can be confusing (and even wrong.)

This has surfaced after my recent changes making true heading more important than magnetic heading, and visible in the info dialog. The reason for that was the added support for GNSS compasses that only produce true heading.

A further complication is that the code receives heading both from OpenCPN (which derives it from NMEA inputs) and from the parsed NMEA sentences, as well as the radar (which does its own derivation.)

A further complication is that it is possible that there are multiple NMEA data sources producing heading, and that OpenCPN doesn't allow filtering on source. I have this myself, and need to "fix" this by filtering.

We could add some filtering mechanism, or revert to prioritizing HDM + VAR over HDT. It is probably simpler to ignore the processed HDM and HDT data coming in from OpenCPN and only use the direct NMEA sentences.

canboat, 2016-08-18:
In other words, current situation is that if you have HDM and VAR coming in that it may also show HDT in the Info message box.

canboat, 2016-08-19:
The new algorithm is to have a priority from:

1. Radar true heading
2. Radar magnetic heading (note: only 1 or 2 will actually be present)
3. NMEA true heading
4. NMEA magnetic heading
5. OpenCPN true heading
6. OpenCPN magnetic heading
7. OpenCPN COG

Note that the NMEA filtering for OpenCPN also applies for the sentences passed to the plugin, and that it IS possible to filter according to NMEA 0183 sender in O, as well as on sentence name and sender+sentence (2, 3 or 5 character prefix filtering.)

marcus-aa, 2016-08-19:
Good, just what I was going to suggest!

douwefokkema, 2016-08-21:
Solved, headings show OK, smooth transition from radar heading to NMEA.To be closed.

closed 2016-08-21 by canboat