keesverruijt/BR24radar_pi #211: 10 minute trails are displayed same way as continuous trails

canboat, 2016-08-18:
Reported by @douwefokkema by email.

canboat, 2016-08-18:
I cannot reproduce (using emulator, but that should have zero effect):

[image]

[image]

Looks exactly as designed (considering my trails go from light blue (recent) to dark blue (old)).

douwefokkema, 2016-08-19:
Tested in harbor with more distinct colors:
10 minute trails switch after about 3 min to second color. After that they seem to last forever (now running 1 hour). Except when they disappear, which happens incidentally, though this could have a different cause (dropping heading or position?).

douwefokkema, 2016-08-21:
Solved in 6068d911f. Can be closed.

closed 2016-08-21 by canboat