keesverruijt/BR24radar_pi #219: Feature: Hot switching between true and relative trails

douwefokkema, 2016-08-26:
Feature request: 
Hot switching between true and relative trails
    - both true and relative trails are recorded when trails == on.
    - when switching between either of them the image is ready at the next sweep,
    no new build-up of trails required

Though most of the time I use true trails, relative trails are very useful to judge risk of collision. When the relative trail point right at you, there is danger of collision. Hot switching makes the judgement easy.
A working solution for this feature is at: https://github.com/douwefokkema/BR24radar_pi/tree/switch_trails

rgleason, 2016-08-26:
New Furuno Radars change the color to red for objects that are a danger (approaching). Is that a possibility?

douwefokkema, 2016-08-26:
We do not yet do any analysis of the radar image itself. Maybe Furuno does that, or they use a Doppler radar. I think however that "getting nearer" is not a sufficient condition to mark a target as "dangerous". The hot switching described above is really very handy and only a minor software change, therefor I propose it as an item for a future release.

Hakansv, 2016-08-26:
Rick
Sorry to be reactionary but this is my opinion: I think is a bad idea. If the system can serve me a good ground for the present situation, like different trails, I'd like to be the one judging what's serious or not. If a system, like OCPN is taking over judging it can be really dangerous if I start trust the system and it fails. And it will fail. Sooner or later. A computer system is not more than code anyone wrote. It can be buggy or things not anticipated can/will happen. The trails are really good to support me about what's going on, but I'll still want to be the one in command.
Or - did I misunderstood and overreacted??

rgleason, 2016-08-26:
No and yes. I just asked a question.  Maybe it is dangerous idea depending on computers, but I would rather make that assessment myself.  We're all depending on one another here, because in one way or another we all contribute to the program.  But I get your point. Why make it more complicated? Trails are better than fine for me.

marcus-aa, 2016-08-27:
Th new furuno radars use doppler shift returns to get an instantaneous speed from the target. After correlating own-speed, it can immediately tell which targets are stationary and which are moving over ground, and the latter are painted in red. It does not know if they are dangerous, just if they are moving.

To decide if a target is dangerous would be more of an ARPA feature that calculates CPA on automatically aquired targets. Furuno is very good at that too, the only mfg that offers true ARPA (as opposed to manual MARPA), in the leisure market.

The Marpa using the Navico 3G/4G with my NSE is not very good imo. The question is how much is done in radar vs in plotter, and if parts of this can be used by the plugin in the future, or if a pure software ARPA could be designed.

/ Marcus

rgleason, 2016-08-27:
Hi Marcus, Yes, I meant doppler shift, but doesn't the B&G broadband use this type of technique now in different ways?  Isn't broadband sending multiple frequencies and seeing what comes back?
Actually from a software point of view its kind of like remembering where "it" was and seeing where it is now. I guess the problem is defining or isolating "it" in the image, and all the computing horsepower to keep track of "it".   Some of Sean's gl graphics talents might be useful here, simplifying the image and being able to analyze the few triangles faster.  I suppose this is completely off topic. Sorry.

Hakansv, 2016-08-27:
The hot switch now successfully tested. Works like a charm, you can instantaneously shift between true and relative trails. Very good.

canboat, 2016-08-28:
Merged in master.

closed 2016-08-28 by canboat