keesverruijt/BR24radar_pi #223: Improve performance of the recording of true trails

douwefokkema, 2016-08-31:
Currently recording of true trails causes an additional CPU load of about 10 - 20 %. Apart from that displaying the trails may cause additional CPU load when in vertex mode (with full displays the shader performs better), but the display operation is not part of this issue.
The recording performance of trails should be improved for small systems and to enable the continues recording of trails.
A possible solution for this and issue 222 is proposed in:
https://github.com/douwefokkema/BR24radar_pi/tree/trails_performance
This solution is not yet tested at sea.

marcus-aa, 2016-08-31:
Maybe we need a setting in the main plugin part for Trail recording on/off. 

If its on, trails are always recorded in background, if off, no trail functions are active at all, for slower systems.

It is the safe way to do it, you either have full trail capability or none.

/ Marcus

douwefokkema, 2016-08-31:
With the performance update that I propose, there is no need to turn off the recording of trails anymore, the additional load of the trail recording can not be noticed.
It sounds like a good idea to have a separate button show / hide trails. Only issue may be that we get too many buttons. What if we use the "true / relative" button and make it into a 3 way button: "true / relative / hide" trails?

rgleason, 2016-08-31:
Is this where Sean's idea would help?

Hakansv, 2016-08-31:
re: "What if we use the "true / relative" button and make it into a 3 way button: "true / relative / hide" trails? "
Yes - That's my own thoughts before your answer! Especially if the CPU load is not more then minor  affected it must be a very good solution. Then the increase and decrease, to off, control can be as it is. 

And, not necessary but a nice to have, could the true/relative/hide button value switch to Relative or True if on Hide when trails time value is changed from Off to any other value.

douwefokkema, 2016-08-31:
"could the true/relative/hide button value switch to Relative or True if on Hide when trails time value is changed from Off to any other value."
Could be done, but should it be True or Relative. To remember the last used seems too much.....
May be better to remove the Off from the time adjustment. So then there would be a button to adjust the trails length and a True / Relative / Hide button

Hakansv, 2016-08-31:
If it's not a performance issue I do agree. 
I haven't checked CPU load for trails as you, so you're the master here. The base for my suggestion was that trails off means no extra CPU load and trails on does. But if you have verified no extra or no significant extra load from constant trails I'm on.

marcus-aa, 2016-09-05:
The max setting of trail length is continous or indefinite. That means that to be able to background record to allow this setting to be hot-switched, the amount of data needed has no limit. Surely that must take some toll on the performance, and definately on storage space. What is a reasonable max time until we discard old trails in the background process ? I suggest 12 minutes.

I'm not a great fan of 3-way buttons. Consider this; a user has relative trails and hides them, then switches back on, only now they are true instead, and it's easy to miss that the presentation has changed.
On the other hand, we are getting into a rather complicated menu systems with lots of buttons and sub-menues. The downside of many nice features....

>And, not necessary but a nice to have, could the true/relative/hide button value switch to >Relative or True if on Hide when trails time value is changed from Off to any other value.

Better not. If we don't need the off option in the length setting it's better to remove it. One setting should not affect another, there is a specific warning about this in the S52 ECDIS standard. Sometimes it's also called "the rule of least surprise..."

canboat, 2016-09-26:
@marcus-aa wrote: 

> The max setting of trail length is continous or indefinite. That means that to be able to background record to allow this setting to be hot-switched, the amount of data needed has no limit. Surely that must take some toll on the performance, and definately on storage space.

Heh, actually no. We store trails very efficiently in a constant space, by continuously increasing a single byte value on each scan. Once it hits maximum age, we keep it there. So 0-255 revolutions is all the same cost, and this imposes a limit of 255 * 2.5 = 637 seconds, or slightly over 10 minutes.

canboat, 2016-09-26:
I say we close this, as the performance is now improved, and open new issues if we don't like the user interface as it is now after Douwe's changes.

closed 2017-08-08 by canboat