keesverruijt/BR24radar_pi #224: 2.1 crashes OCPN

marcus-aa, 2016-09-05:
2.1 has a bug that crashes ocpn (or ocpn has a bug that makes it crash when radar_pi sends some specific data to it)
2.0 is reportedly working ( my friend also has the crash problem on 2.1 but not on 2.0, using a 3G and same GPS compass).

- Emulator mode is fine 
- Stand-alone radar is fine, without compass or GPS connected
- With GPS and HDT, it crashes immediately when radar A is leaving standby
- With GPS and HDT, radar B can be started, even with overlay display, but crashes after a while

I suspect it has something to do with feeding the heading data from the radar to ocpn, maybe only affecting true heading ?
Both our installations has 2 sources of HDT/HDM, one from NMEA and one from the radar, from different compasses, this  could be related. I can sometimes get ocpn to crash if hot-switch HDT source, even without radar running....

douwefokkema, 2016-09-06:
Can  you make a (or a few) logfile(s) with verbose = 15, maybe that gives an indication. I have not seen a crash on Windows, so I can't reproduce.

canboat, 2016-09-06:
Can you check whether you have "Pass radar heading to OpenCPN" set? If so, does it still reproduce if you disable this?

canboat, 2016-09-06:
I tried reproducing on Ubuntu x86_64 16.04, but no luck.

marcus-aa, 2016-09-06:
It seems related to HDT in from nmea. I will sort out the log files and make available, but all crashes have this HDT sentence in common:

23:29:41 CEST: BR24radar_pi: SetNMEASentence $APHDT,357.5,M*2E

It doesn't matter if  "Pass radar heading to OpenCPN" is on or off.
Is this the sentence that radar_pi  generates and loops back into ocpn ? There is no other "AP" talker on the NMEA inputs. 

It stops crashing if we set a NMEA input filter that blocks HDT...!

[image]

Hakansv, 2016-09-07:
Strange......
1)
Your HDT message says: ...HDT,353,5,M*..... But the official HDT sentence don't have the "M" option only "T". If "M" the message should have been called HDM. 
2) I haven't checked how this inconsistency would affect BR24 behavior. Kees can you make a check?
3) Your filtered, magenta, HDT message is from the IP connection. From where is "Virtual" HDT comming? BR24?? (Could be checked by deactivating BR24.

canboat, 2016-09-07:
We send HDT,xxx.x,M from PassHeadingToOpenCPN(). This is wrong, obviously.

I think "Virtual" is coming from the BR24radar_pi, but that doesn't explain why it is also coming from UDP. Unless this is a bug in OpenCPN.

I think what is happening is that the HDT sentence loops back from BR24radar_pi to O and then back into ourselves, and that the (imported) NMEA code croaks on the HDT message. Weird.

Fixing it to send ,T is simple, and we can also just ignore all incoming Heading messages if we already have heading from radar. But I'd rather find out what the real cause is.

@marcus-aa if you're able to compile the plugin yourself, and compile a debug version, you can then run opencpn as follows:

     gdb opencpn

and it will generate a nice stacktrace when it crashes (dumps core, as I like to call it. I actually *owned* a core based computer, once, a PDP8.)

marcus-aa, 2016-09-07:
I'm away from the boat now for 3 weeks, gone hunting up north. The logs are from my friend  with similar setup and 3G radar. I can test when back but he needs compiled binaries for testing.

The UDP input comes from another computer with ocpn, and possibly it is the radar_pi running in the background there producing this APHDT sentence. He can check this by disabling plugin later.

There seems to be several parts of this problem (as often with these things).

1. The APHDT sentence is malformatted with "M"
  (What is the reason for the AP talker ID ? Wouldnt RA,HD or EC be more logical?)

2. radar_pi sends this sentence even if "Pass heading" is off, and also even if no compass is connected to radar... !

3. Possibly radar_pi remembers an old hdt and keeps sending this, we saw 055 deg when  boat is almost at 000.

4. Ocpn chokes on a malformatted NMEA sentence, which is also strange/ not good .

Here are the log files:

Normal settings: Pass Radar heading = ON, HDM compass into ocpn by NMEA (IIHDM), and HDT compass connected directly to radar by N2K to RI10.
https://www.dropbox.com/s/d50p5fnvc5tgqhd/Passhdg_on_hdt_on_hdm_on.log?dl=0

Pass Radar heading = OFF, HDM compass into ocpn by NMEA (IIHDM), and HDT compass connected directly to radar by N2K to RI10.
https://www.dropbox.com/s/d50p5fnvc5tgqhd/Passhdg_on_hdt_on_hdm_on.log?dl=0

Pass Radar heading = OFF, HDM compass into ocpn by NMEA (IIHDM), and NO compass connected directly to radar.
https://www.dropbox.com/s/k17nnigkb6hbh1h/Passhdg_off_hdt_off_hdm_on.log?dl=0

Pass Radar heading = OFF, HDM compass disconnected, HDT compass disconnected, COG as heading, overlay off, and APHDT sentence blocked in NMEA filter.
https://www.dropbox.com/s/nf3ct63hr0zbv1o/Passhdg_off_hdt_off_hdm_off_coghgd_on_overlay_off_HDTsentence%20blocked.log?dl=0

In the last example the radar runs for a bit, but after a while O crashes anyways, with a APHDT sentence in the log file...

douwefokkema, 2016-09-08:
I examined the logs. Tried to reproduce by sending same wrong message ($APHDT,357.5,M*2E) to OCPN with a serial tool. Reproduced the crash one time, but not again after that. Did not get more info out of that. 
" radar_pi sends this sentence even if "Pass heading" is off, and also even if no compass is connected to radar... !" Did not find that, no messages send with "Pass heading" is OFF. 
" The APHDT sentence is malformatted with "M"" This is corrected in my latest version at https://github.com/douwefokkema/BR24radar_pi/tree/trails_performance but for linux you will need to compile.

douwefokkema, 2016-09-08:
Can reproduce now, Just wait long enough. Analysis: this NMEA messages causes a change in heading source, leading to the log message call: LOG_INFO(wxT("BR24radar_pi: Heading source is now HDT %d from NMEA %s (%d->%d)"), m_hdt, sentence.c_str(), m_heading_source, HEADING_NMEA_HDT); 
This call is issued from a secondary thread, and wxWidgets is not thread proof, causing a crash. Bug in wxWidgets! I pushed a solution in the above mentioned branch. That branch does however contain many other updates that are under test.
Don't forget to set VerboseLog=0 again in the ini file!

canboat, 2016-09-08:
I'm doubtful this is it, but we can try. Wxwidgets 2.9.5 upwards the logging should be thread safe. 

I won't be able to check this until the 18th, I am going sailing.

douwefokkema, 2016-09-08:
It is running for hours now. But with the LOG_INFO it does crash within 10 min, time after time. I did make detailed logging, it crashes exactly at the LOG_INFO (shortLOG_INFO's don't crash) . Same with 1.3, it only became fully stable after removal of all wxLogMessage from the receive thread.

Hakansv, 2016-09-08:
Out sailing. I'll make a live test tomorrow. But one note so far:
NMEA messages contain chars for a new line and a new row so the log message for me looks like this:

21:30:55: BR24radar_pi: Heading source is now HDM 212.000000 + VAR 3.141307 from NMEA $IIHDG,212.0,0,E,0.0,E*56

 (2->5)
21:30:55: Chart cache purge

Hakansv, 2016-09-10:
I used the last from trails_performance for some hours with transmitting radar and got one OCPN crash/hang but I was sailing and didn't had time for further investigations. No significant log entry before the crash.

marcus-aa, 2016-09-26:
I have now managed to compile my first code on Ubuntu, wasn't too hard. I used the https://github.com/douwefokkema/BR24radar_pi/tree/trails_performance as Douwe suggested. It seems to fix the HDT-related crashes. 

As it is a branch of 1.3 it has some other problems, like applying variation to true heading and switching back and forth between heading sources, but that should be OK if the log fix is applied to the 2.x code ? 

It also froze the radar presentation when trying to correct bearing alignment and one other time when working the settings, but i suppose it's no point in making issues for the 1.3 branch, better wait for a fix in 2.x code so I can test the current release+fix instead?

canboat, 2016-09-26:
@marcus-aa it seems you compiled the wrong branch, Douwe's trail_performance branch is also 2.x based and it only has very small changes compared to the last 2.1 release.

marcus-aa, 2016-09-26:
Great, 2 hours wasted.... :-( At least I learned how to compile.

I follow the link, make sure branch = trails performance, click the green "Clone or download" button, but it serves up the default https://github.com/douwefokkema/BR24radar_pi.git link (is that "master" ?) regardless of which branch is selected. 

Am I stupid or is Github broken... ? :-))

How do I get the code from a branch ?

Hakansv, 2016-09-26:
I don't now what tool you use for Git but if in the command window:

If you're on your master I suggest to first make a new branch, why not 
same name. To make a new branch from master:
Git checkout -b trails_performance

Then get Douwe's code to that branch:
Git pull https://github.com/douwefokkema/BR24radar_pi trails_performance
One space between the url and the branch name.

Håkan

Den 2016-09-26 kl. 16:28, skrev Marcus AAkesson:

canboat, 2016-09-26:
@marcus-aa, here is an alternative to what @Hakansv wrote.

I think you did:

```
$ git clone https://github.com/douwefokkema/BR24radar_pi.git
Cloning into 'BR24radar_pi'...
remote: Counting objects: 6220, done.
remote: Compressing objects: 100% (57/57), done.
remote: Total 6220 (delta 38), reused 0 (delta 0), pack-reused 6163
Receiving objects: 100% (6220/6220), 19.78 MiB | 6.32 MiB/s, done.
Resolving deltas: 100% (4870/4870), done.
Checking connectivity... done.

```
To switch to the trails_performance branch:

```
$ cd BR24radar_pi/
$ git branch
* master
$ git checkout trails_performance
Branch trails_performance set up to track remote branch trails_performance from origin.
Switched to a new branch 'trails_performance'

```

douwefokkema, 2016-09-26:
I now have a new branch "all" (https://github.com/douwefokkema/BR24radar_pi/tree/all) that contains also the course_up orientation.

canboat, 2016-09-26:
And I merged Douwe's "all" branch into canboat/master just now.

marcus-aa, 2016-09-26:
Thanks for the help!

I didn't checkout anything, I just followed the help and on the target machine did

git clone <pasted link> 

That link was provided by the green Github download button. But that button is broken, even when a branch is selected it does not reflect that.

So I should have manually added the branch after that link ?

git clone https://github.com/douwefokkema/BR24radar_pi trails_performance 

would that have worked ? I'm not at the boat, don't know when i can test next time.

canboat, 2016-09-26:
As I have now merged, I suggest you just use

    git clone https://github.com/canboat/BR24radar_pi

marcus-aa, 2016-09-30:
Compiled 2.2 and it worked without crashes, just testing 5 minute at the dock. Will try to make a real test on Sunday if weather permits.

canboat, 2017-01-15:
Closing, presumed fixed. Just re-open this if it happens again.

closed 2017-01-15 by canboat