keesverruijt/BR24radar_pi #225: Overzoom when changing range

marcus-aa, 2016-09-05:
When changing range using mouse scroll OR buttons in Radar windows, it seems to overshoot the zoom level, going first too far then immediately back to the correct display.

canboat, 2016-09-07:
It's a visual artifact, we change the range internally but don't draw the old spokes correctly. We should zoom the spoke memory when the range is changed. An alternative is to erase the entire screen.

canboat, 2017-08-15:
I fixed a number of issues that were related to this, but it is not perfect yet. In particular the ARPA contours can get drawn at the wrong distance.

canboat, 2017-08-15:
Fixes include today's commit f0f04cb44c3d3aaac0fc18d5d3d3bfec2e2a282f for fixing the shader -- it never fully erased the picture when a range zoom was done.