keesverruijt/BR24radar_pi #227: Suggest adding trail duration 6min & 12 min

marcus-aa, 2016-09-05:
6 minutes is probably the most common prediction vector length and it would be good to have the same time on trails.

(3,6 & 12 minutes are very logical in a sexagesimal time system, much more so than 5 and 10 minutes.)

canboat, 2016-09-07:
I'm happy removing 5 and 10 and adding 6 + 12 instead.

douwefokkema, 2016-09-15:
I implemented the 6 +12 min trails. And it does not work properly. Problem is that we count the number of revolutions for the trail duration. 12 Minutes corresponds to 288 revs. This is larger than 256 and does not fit in a single byte. Internally all trail manipulations are based on single byte values. Changing this not worth the effort and additional memory  resources. Originally we had 3 min and 10 min, now we have 3, 5 and 10 min, should be sufficient.

closed 2016-09-15 by douwefokkema