keesverruijt/BR24radar_pi #228: No way to set true cursor bearings when in head up mode ?

marcus-aa, 2016-09-05:
If heading is available, and correctly displayed on the outer heading circle in the radar window, I would prefer if cursor bearings are also true.

Only use relative bearings if no heading available? Or do we need another button....

Or have I just not found it... ?

canboat, 2016-09-06:
Yes, at the moment we show relative beharing in head up mode. This is what the Navico plotter does. Show both? Preferences page is another possibility. 

Met vriendelijke groeten,
Kees

canboat, 2016-09-26:
The new course up mode developed by Douwe will give you true bearings. Head up mode is meant for users without a compass now.

canboat, 2017-01-15:
@marcus-aa is the course up mode good enough? If so please close, otherwise comment what your reasoning is.

canboat, 2017-08-15:
Heading up mode is now replaced by Heading up (Stabilized) when you have a compass heading, and this shows bearings in true degrees.

closed 2017-08-15 by canboat